ramapcsx2/gbs-control #272: Could not compile sketch: "call of overloaded 'abs(unit32-t)' is ambiguous" in framesync.h

Vendest, 2021-09-22:
Error message during verify/compile

In framesync.h

In file included from C:\Users\admin\Desktop\gbs-control\gbs-control.ino:20:
C:\Users\admin\Desktop\gbs-control\framesync.h: In static member function 'static bool FrameSyncManager<GBS, Attrs>::findBestHTotal(uint32_t&)':
framesync.h:148:33: error: call of overloaded 'abs(uint32_t)' is ambiguous
  148 |     if (abs(inPeriod - outPeriod) <= 4) {
      |                                 ^
In file included from c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\cstdlib:75,
                 from c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\stdlib.h:36,
                 from C:\Users\admin\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:27,
                 from sketch\gbs-control.ino.cpp:1:
c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\stdlib.h:74:5: note: candidate: 'int abs(int)'
   74 | int abs (int);
      |     ^~~
In file included from c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\cstdlib:77,
                 from c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\stdlib.h:36,
                 from C:\Users\admin\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:27,
                 from sketch\gbs-control.ino.cpp:1:
c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:79:3: note: candidate: 'constexpr long double std::abs(long double)'
   79 |   abs(long double __x)
      |   ^~~
c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:75:3: note: candidate: 'constexpr float std::abs(float)'
   75 |   abs(float __x)
      |   ^~~
c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:71:3: note: candidate: 'constexpr double std::abs(double)'
   71 |   abs(double __x)
      |   ^~~
c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:61:3: note: candidate: 'long long int std::abs(long long int)'
   61 |   abs(long long __x) { return __builtin_llabs (__x); }
      |   ^~~
c:\users\admin\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:56:3: note: candidate: 'long int std::abs(long int)'
   56 |   abs(long __i) { return __builtin_labs(__i); }
      |   ^~~
exit status 1
call of overloaded 'abs(uint32_t)' is ambiguous

askotx, 2021-09-23:
Use an older commit of this repo -at least previous commit- and test again, that' s what I did, also you can add ESPAsyncTCP and ESPAsyncWebServer newer repos.

Vendest, 2021-09-28:
Problem solved. I did not read the instruction carefully when adding board esp8266 in board manager. 
There is a note "Do not use any version >= 3.0.0, it is reported to break things!:" indeed it does.
Thanks

closed 2021-09-28 by Vendest