ramapcsx2/gbs-control #274: Cannot update firmware again

rabidgerry, 2021-09-24:
Hi all, I had a similar problem a few months back and I believe it was related to my esp8266 library being too new.  So I downgraded and from what I remember it worked no problem.  So I was trying to update the firmware again and today got another problem (see below).

For the record I uninstalled Arduino v1.8.15 and then re-installed.

And also reinstalled all the libraries again just in case.  Despite being able to compile, I was not able to upload and I got the following errors below.  Can anyone help?  Thanks in advance.

**Traceback (most recent call last):
  File "C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3/tools/elf2bin.py", line 182, in <module>
    sys.exit(main())
  File "C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3/tools/elf2bin.py", line 166, in main
    to_addr=4096
  File "C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3/tools/elf2bin.py", line 161, in wrapper
    write_bin(out=out, args=args, **kwargs)
  File "C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3/tools/elf2bin.py", line 75, in write_bin
    [size, addr] = get_segment_size_addr(elf, segment, args.path)
  File "C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3/tools/elf2bin.py", line 55, in get_segment_size_addr
    raise Exception('Unable to find size and start point in file "' + elf + '" for "' + segment + '"')
Exception: Unable to find size and start point in file "C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.3/bootloaders/eboot/eboot.elf" for ".text"
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).**

rabidgerry, 2021-09-24:
I tried to go back to the old firmware and upload again and got this

rabidgerry, 2021-09-24:
**Arduino: 1.8.15 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from C:\Users\theme\Documents\gbs-control\gbs-control.ino:20:

C:\Users\theme\Documents\gbs-control\framesync.h: In static member function 'static bool FrameSyncManager<GBS, Attrs>::findBestHTotal(uint32_t&)':

framesync.h:148:33: error: call of overloaded 'abs(uint32_t)' is ambiguous

  148 |     if (abs(inPeriod - outPeriod) <= 4) {

      |                                 ^

In file included from c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\cstdlib:75,

                 from c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\stdlib.h:36,

                 from C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:27,

                 from sketch\gbs-control.ino.cpp:1:

c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\stdlib.h:74:5: note: candidate: 'int abs(int)'

   74 | int abs (int);

      |     ^~~

In file included from c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\cstdlib:77,

                 from c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\stdlib.h:36,

                 from C:\Users\theme\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Arduino.h:27,

                 from sketch\gbs-control.ino.cpp:1:

c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:79:3: note: candidate: 'constexpr long double std::abs(long double)'

   79 |   abs(long double __x)

      |   ^~~

c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:75:3: note: candidate: 'constexpr float std::abs(float)'

   75 |   abs(float __x)

      |   ^~~

c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:71:3: note: candidate: 'constexpr double std::abs(double)'

   71 |   abs(double __x)

      |   ^~~

c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:61:3: note: candidate: 'long long int std::abs(long long int)'

   61 |   abs(long long __x) { return __builtin_llabs (__x); }

      |   ^~~

c:\users\theme\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits\std_abs.h:56:3: note: candidate: 'long int std::abs(long int)'

   56 |   abs(long __i) { return __builtin_labs(__i); }

      |   ^~~

exit status 1

call of overloaded 'abs(uint32_t)' is ambiguous

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.**

Robertcop84, 2021-10-05:
I have the same. I just wanted to update to a wemos d1 mini board and got stuck right there.

Robertcop84, 2021-10-05:
Quote from Software Setup guide

Go to "Tools" > "Board:" > "Boards Manager ...", search for "esp8266" and install "esp8266 by ESP8266 Community",
version 2.6.3 (or newer) EDIT: Do not use any version >= 3.0.0, it is reported to break things!

Have fun!

rabidgerry, 2021-10-05:
> Quote from Software Setup guide
> 
> Go to "Tools" > "Board:" > "Boards Manager ...", search for "esp8266" and install "esp8266 by ESP8266 Community", version 2.6.3 (or newer) EDIT: Do not use any version >= 3.0.0, it is reported to break things!
> 
> Have fun!

Sorry not sure what your suggesting here??  Every time I go in update my GBSC I end up following the guide so I've followed those steps to a "T" and still no luck.  The esp8266 version I have is 2.6.3.  I did have 2.6.4 but downgrading further to 2.6.3 didn't do anything either.  Pretty sure I updated just about the last time using 2.6.4.

You have been successful then?

Robertcop84, 2021-10-05:
Yes I have with 2.7.4