ramapcsx2/gbs-control #276: The buttons on the GBS8200 don't do anything.

GrumpyModeler, 2021-09-29:
The buttons on my GBS8200 don't seem to do anything. It would be practical to use the up and down buttons to change between presets. Would that be possible? IS there a way to navigate between presets without logging in to the web interface  that I'm missing?

askotx, 2021-09-30:
When you put a jumper on the board, the esp8266 take control over the image and as far as I know, not over the hardware.
About your last question, I think the answer it's no.

GrumpyModeler, 2021-09-30:
I guess it would be possible to use two of the GPIOs for buttons to swap profiles.

TABYDACAT, 2021-10-01:
Going to chime in for this. GBS Control has an optional way to do this already. Yes, for your question, the OLED system is kinda overkill, but the feature you're asking about is there already. While it is limited to a few presets, it's possible to store a few and select them without the web interface. 

Second question, technically, just transfer those buttons to the GPIOs, write some code to switch between the profiles. Like anything else, this comes with a set of constraints, but try it out and form it to fit your needs. 

Good Luck!