ramapcsx2/gbs-control #277: Replacing C23 C41 C42 C48 instead of stacking extra capacitors

karas84, 2021-10-02:
So I was wondering what capacitor should I use if I wanted to replace the stock C23 C41 C42 C48 instead of adding an extra one on top of them. I tried to search for schemathics but wasn't able to find any.

Also, reading the forums I found out that, since C41 can be difficult to operate with, it is possible to replace only C43. Is it correct? If yes, should I replace C43 with a bigger (double capacitance) capacitor or just one is fine?

GXTX, 2021-10-05:
Mmm docs need updating, really no reason to be stacking caps.

karas84, 2021-10-05:
Are you sure? According to @ramapcsx2 (both in the readme and in the forums) stacking these caps makes a good difference in terms of image noise filtering.

GXTX, 2021-10-05:
There's no reason to stack the caps instead of replacing them.

ramapcsx2, 2021-10-07:
You can stack or replace. Stacking is preferred, since you combine the benefits of 2 capacitors that way.

cbaecchi, 2021-10-11:
@ramapcsx2 Very interesting video, thanks. I think I now understand why stacking is better than simply replacing the caps.

closed 2021-10-11 by ramapcsx2

cbaecchi, 2021-11-01:
Yeah, things get crazy reeeeeeally fast when moving from theory to real-life :D. Thanks to you for sharing it and for the effort you put into gbs-control!

karas84, 2021-11-01:
Yeah, things get crazy reeeeeeally fast when moving from theory to real-life :D. Thanks to you for sharing it and for the effort you put into gbs-control!