ramapcsx2/gbs-control #280: Greyed out (unselectable) outputs etc...

segacollection, 2021-10-11:
I'm not able to select various outputs (or other options) on the GBS-C but until recently it had been working fine. Is it input and output cable dependent? I ask as I am, for the first time, trying to go from 480p to 15 kHz. I'm sure I've been able to select this before, but right now only 960p is an output option (not even 720p).

Any comments or tips?

segacollection, 2021-10-11:
Made progress but still a bad /unsynced picture. Any thoughts appreciated. 

[image]

ramapcsx2, 2021-10-11:
Hm, the logs say you're outputting a high resolution preset?
A regular TV will only work with downscaling, or an already 15kHz source in passthrough.

segacollection, 2021-10-11:
This is what I get when I click 15 kHz (input is now a PS4 in 480p mode) 

[image]

segacollection, 2021-10-18:
Hmm. I use one of these little HDMI-VGA DAC things from Amazon, so I guess not. I'll have a look into the sync combiner thing.