ramapcsx2/gbs-control #290: A lot of noise and "vertical flicker" after installation

dabilde, 2021-11-01:
I'm pretty sure I got a bad GBS-8200 board. But before I buy a new one and try again, I wanted to know if somehow there's a fix for this. Here's what I did: soldered 5 wires from the GBS to the ESP8266, 5 wires from the GBS to the clock gen, bridged the jumper and removed C11 (which didn't help, I'm pretty sure). I didn't do any other mods cause I'm not planning on using a scart input, I only want to use component. The web UI works fine and the clock gen is detected as reported by the developer console. I checked and rechecked all solder joints. Here's a video of the issue: https://www.youtube.com/watch?v=1vIKY3BiFrw

Thanks in advance.

Nuck-TH, 2021-11-01:
You can start by making wires between ESP board and GBS shorter.

dabilde, 2021-11-01:
> You can start by making wires between ESP board and GBS shorter.

Thanks for the tip, just made them a lot shorter. Didn't solve it, though.

tomudo, 2021-11-01:
What is the cause you think  your GBS-8200 board is bad?

Do you try to use factory default (not GBS-control)?

dabilde, 2021-11-01:
> What is the cause you think your GBS-8200 board is bad?
> 
> Do you try to use factory default (not GBS-control)?

Yeah, I tested it once without the mods and the flicker was there but I thought the installation of the clock generator would fix it.

ramapcsx2, 2021-11-01:
Looks like bad SDRAM on the GBS, or a related issue.

dabilde, 2021-11-01:
> Looks like bad SDRAM on the GBS, or a related issue.

So I should just try to get a new one, right?

ramapcsx2, 2021-11-02:
I think a new one would quick fix the problem, but you can try finding out what the problem with your current board is:
In the web ui (development tab, I think), try setting a different SDRAM clock, see whether it helps.

dabilde, 2021-11-03:
> I think a new one would quick fix the problem, but you can try finding out what the problem with your current board is: In the web ui (development tab, I think), try setting a different SDRAM clock, see whether it helps.

Bought a new one, and it was still happening so I got to troubleshooting.

Turns out it was the AC adapter! I bought one from a spanish website called ArcadeXpress, (5v 2A), that they sell with GBS boards, and I guess it's faulty. I didn't have another adapter so I used a 5v 2A usb power brick connected to the ESP's micro usb port and it's working perfectly (is this safe to use?).

ramapcsx2, 2021-11-03:
You must power the GBS through its voltage regulator, otherwise the load is too heavy.

dabilde, 2021-11-04:
> You must power the GBS through its voltage regulator, otherwise the load is too heavy.

Interesting. The header on the GBS that I'm using to connect to VIN and GND on the ESP has "DC 5-12V IN" written just under it. Doesn't this mean that it's still going through the regulator after the header?

ramapcsx2, 2021-11-04:
Okay, the power hierarchy should be: 5-9V into the GBS, then either 3.3V out of the GBS into the ESP or make the power input parallel with the ESP (but then limit to 5V). 
It is just important to power the GBS by its own regulator. Secondary is the ESP, which should just not be fed with too high of an input voltage into its own regulator.

The barrel plug on the GBS is the same connection as the header right next to it, if you mean that.

dabilde, 2021-11-04:
Got it. Thanks for all the help.

closed 2021-11-04 by dabilde