ramapcsx2/gbs-control #291: GBS board not responding

BuddiScotch, 2021-11-03:
Hi All,

To start off, yes the SDA and SCL pins have been triple checked.
I've tried multiple power supplies.
I've tried reflowing the solder on the contact points.
I've restarted it and reset all defaults.
I've tried switching between access point and connecting to my own wifi to remote in.
I've tried different VGA to HDMI adapters.

I'm sure I'm missing a bunch of basic troubleshooting steps that I absolutely did but don't remember. You know the drill.

If you've made it this far I place my future in your hands.

So my GBS board keeps throwing me the error of "GBS board not responding!" whenever I try to select a resolution. All the resolution options are greyed out.

It does not display anything, and on my TV it has no signal. So its not receiving any signal from the GBS 

Something to keep in mind is that the GBS board did not have any issues *prior* to me soldering the clock generator to the board. Afterwards, it was was having these issues but for a short period of time it still worked. However now its completely dead.

The wifi board still pops on the blue light when I boot, however it does not come on after that.
That and the red power LED for the board works.

I believe that's everything.

Thanks Al
[image]
[image]
[image]
l
[image]
[image]
.

TABYDACAT, 2021-11-03:
Have you tried utilizing C41 as well just to troubleshoot the clock gen?

closed 2021-12-23 by BuddiScotch

israpps, 2023-03-13:
@BuddiScotch Hi

im having this issue after clockgen install.

how did you fix the issue?