ramapcsx2/gbs-control #292: Green Haze feedback

shawneeunion, 2021-11-03:
Really enjoyed the GBS-Control but have had a lingering issue with a green haze on the screen. I removed the pots and shorted the connections. I also have tested different component cables (Wii and PS2) and the issue persists. Also tested turning off/on ADC and it slightly lessens the issue when off. Using the GBSC with TCL 6 series tv with most settings set to normal. 

[image]

GrumpyModeler, 2021-11-03:
Have you tried other video signal cables?

shawneeunion, 2021-11-03:
Good too. I tried the GBSC with a different hdmi cable and noticed pink horizontal flickering on the screen. Read some other comments and saw that shortening the wires may help. Going to see if that fixes it.

TABYDACAT, 2021-11-04:
How are you outputting the signal VGA or a converter of some sort?

shawneeunion, 2021-11-04:
Vga to HDMI this specific board has two vga outputs.

tomudo, 2021-11-04:
I have similar color issue with Wii 3 wires component video output. You may try toggle "IF Auto offset" in developer tab for temporary fix.

shawneeunion, 2021-11-04:
Thanks @tomudo . Going to to try that once I square away the new issues. 

I ended up shortening the wires and now it giving me the, “Please check power and cabling” warning in the logs. I’m suspecting my ESP8266 is a faulty clone. It played for about 45 minutes and just froze before being unresponsive. Checked the voltage to the ESP and it was 5v and there is continuity on all the connections. 

Previously, I haven’t been able to access the GBSC web interface while video is processing. Interface works fine if no video is being processed but once signal is sent from the console, PS2, to the device the web interface just 404s. Wondering if there is anything else to check before grabbing a new ESP.

ramapcsx2, 2021-11-04:
The picture you took looks like either it is an offset issue in the GBS, or the PS2 is set to the wrong output mode in its BIOS.
If it's the GBS, try the "IF Auto offset" option.

ramapcsx2, 2021-11-04:
Unresponsive WebUI is a known problem, where the scaler interferes with the small WiFi antenna. It can go away with different consoles, different resolutions and such.

shawneeunion, 2021-11-04:
Going to try the “if auto offset” option soo. Removed power to the ESP8266, going to try a Wemos D1 and follow up. GBS with on board controller gives this output when upscaled.

[image]

EmmerichFrog, 2021-11-08:
@ramapcsx2 I have a similar issue, on mine the green tint it's not as bad and it goes away completely when I toggle IF Auto offset. My only issue is that the setting doesn't seem to save to a preset, so I need to toggle the option every time. Am I doing something wrong or is it normal that it resets every time?

shawneeunion, 2021-11-28:
Wanted to give an update. I think this green haze has more to do with my TV than the GBS Control. It's a TCL 6 Series, I had setup a new GBS control using an 8200 rather than an 8220 and the results were very similar. I then noticed playing some games on Retropie that the TV would auto-adjust dark portions of the screen to the green haze. Not sure what setting does this but it does not happen on my standard computer monitor.

closed 2021-11-28 by shawneeunion