ramapcsx2/gbs-control #295: I can't get GBSControl to work with my Amstrad CPC 464

GrumpyModeler, 2021-11-14:
I have tried using GBSControl with GBS 8200, 8220 and HD 9900 with my Amstrad CPC 464 with the same result. It refuses to lock on. 

Retrotink SCART handles the signal from this computer fine, and so does my LCD TV and several CRT TVs I have tested. I think it worked fine on my OSSC too when I tried that a while back (my OSSC is used on my consoles).

I have tried these things, which did not fix it: 
-Including a proper LM1881 circuit. Tried with and without V-sync output.
-Sync from Luma and sync from C-sync. Same result from both, including when I run it through an LM1881.
-Changing C33 and C35 to 1nF, 470 pF, 2nF and 47 nF capacitors. 47 nF gives the closest to working result, which is what you see in the video below.
-All settings on and off in the options in GBSControl.
-Moving the picture around and scaling in GBSControl.
-Removing the 100 Ohm resistor mod between the sync and ground.

I think the CPC just has slightly unusual video modes.

Here's how it behaves:
https://youtu.be/t1TSSc7JTUc

For now I'm resigned to using my RGB SCART Retrotink, which isn't great as scaling and centering isn't possible on my monitor or on the Retrotink.

Matty72, 2022-01-06:
CPC has inverted HSync, possibly the issue, possibly something that gps-control can address?
My GBS8200 has a companion CGA to analog converter board that cleans and allows inverting the Syncs. Prior to gps-control mod, my CPC would not display correctly until i inverted tbe HSync (CSync from CPC fed to HSync).

Matty72, 2022-01-06:
Oh, refer below for the CGA to Analog GBS8200 companion board - it has a digital/analog input switch so works with non-CGA analog inputs like the CPC, plus a C64 color correction option(yellow to brown - i think IBM CGA did the same)

http://www.microbeetechnology.com.au/store/cga-to-analogue-adapter-board-for-video-scalers.html

GrumpyModeler, 2022-01-06:
I didn't think that mattered to GBS Control. I'll check if I have any inverter IC's and see if that helps.