ramapcsx2/gbs-control #296: GBS v5.1 unable to use GUI

Slypty, 2021-11-17:
So when I access the gbscontrol with the v5.1 board, if I press any of the options they don't respond.  Instead, I get a flashing red wifi signal on the top right of the gbscontrol page.   Basically saying 'I have no connection.'.  I've talked to another modder with this version, and he's having the same issue.  I noticed that on my phone, the best I could get is to have the settings jump from pass through to one other upscaled resolution, but it wasn't what I had pressed as my selection.  I've tried 3 different computers for the access point, and my phone is the most responsive.
  
   Also, once I connected to my home network, the gbscontrol won't load on my home network.  I've tried using all the different access point names, but none work.  I'm new to the WiFi setup so I may be overlooking something simple.

Gabriel2Silva, 2021-11-23:
I have the same v5.1 board and it works fine on both AP and Wi-Fi mode. It's a little unstable, yeah, but when it doesn't work I simply power cycle it. Check your routers client list and check for the GBS IP Address. It'll be randomly assigned by your routers DHCP. You can then access it by typing the IP in your browser. 

Something that helped immensely in my experience is to reserve an IP Address to the GBS' MAC Address in your router. That makes it so everytime your GBS tries to connect, your router will give him the same local IP address. After doing this, I didn't have single issue regarding connectivity.

Slypty, 2021-11-24:
Thanks for the tips.  I tried logging in to my modem, but the gbscontrol IP doesn't show up in the DHCP Client list.  When I try to connect to it, I'm able to for about 10 seconds then the connection is lost.  I'm not able to continue after that, I just get logged out and can't select anything.  It's a bit odd my router doesn't see it, but I can log in to it on my Chromebook and my Phone.

GrumpyModeler, 2021-11-27:
Try turning off mobile data (4G or whatever). It's probably trying to connect via mobile data if that's active even if it's connected to the GBScontrol wifi too. I have to turn off mobile data to access the GUI of my GBScontrol boards. This also goes for other things like cameras with WIFI control.

ramapcsx2, 2021-11-27:
Yep, that often helps.

Slypty, 2021-11-29:
That can't be right, the phone I'm using doesnt have a SIM card.  I'd
suspect my ESP8266, but I've had the same results twice with two different
ESP8266s.

GrumpyModeler, 2021-11-29:
I have built several units with V5.1 board using both generic ESP8266 NodeMCU boards and D1 Mini Pro 4M Bytes WEMOS boards without issues for my retro computer room. The generic NodeMCU boards can have pretty bad WIFI reception though because of the on board PCB antenna.

Slypty, 2021-11-30:
I don't t doubt the board can be used, maybe I've overlooked something with the install.  Here's some pics of my install.  Does the blue led light up when you plug it in?  
Mine never seems to.  I've tried running a component signal from my GameCube to the VGA output in to a ViewSonic PC monitor.. and I get no signal either.  I haven't used a GBS ever, so 
I'm not sure if the GBS is at fault, my install, or even my GameCube's compatibility.
[image]
[image]
[image]

GrumpyModeler, 2021-11-30:
Yes the blue led lights up for a short while on all mine when I turn them on. Is it getting power? You have connected the power wire to the 3,3V point on the GBS. Try powering it with 5V instead. It says on your ESP8266 that 5V is recommended. 

I power my GBS with 5V and power the ESP8266 with a wire from the  5V input on the GBS.

Slypty, 2021-12-01:
It is getting power. The pic above doesn't really show where the red wire goes to, but I do have it connected to the VCC pin.  If I was just running Gamecube to the GBS, then output to VGA shouldn't I at least see a picture, WiFi or not WiFi?

[image]

ramapcsx2, 2021-12-01:
When in doubt whether a GBS build works at all, try different source consoles. The Gamecube is known to be hard to sync to.

Slypty, 2021-12-02:
Well the GBS works fine with my Wii (component to VGA).  So I've ordered
another WiFi module, considering it's being powered and I see a WiFI link
for it, the issues persist with selecting any outputs from the firmware.
Basically it won't select anything, and continues to disconnect from the
gbs-c module.

Slypty, 2021-12-16:
Figured it out. Maybe specific to my PC, not sure.  The idea of adding
libraries to Arduino via extracting the .zip to the Libraries folder wasn't
working properly.  The correct method was to go to 'Sketch' and 'Add
Libraries' in the Arduino interface, and use the 'add 'zip to libraries'
option.  This isn't mentioned in the tutorials, but again it could just be
my PC being a bitch.

Matty72, 2022-01-06:
I had same issue, then realised the downloaded zip and the root folder within are being appended with "-master", once removed, no errors upon compile.

bobdillon, 2022-01-22:
I'm confused here. I also have an issue with an unresponsive UI, but how were people able to compile the script without the libraries it requires? 

Did we change subjects to a compilation error in the same thread? 

Or is it somehow possible to upload code missing prerequisite libraries and therefore wind up with a buggy UI as a result? 

Thanks in advance.

Slypty, 2022-01-22:
Another thread hasn't been created, I might post as a compilation error if
the fix works for you.
As mentioned, at least what my issue was that adding the libraries with the
method on the tutorial page for GBS-C did not work for me.  Instead, try
adding the libraries manually (instead of putting the .zips directly in the
library folder.)  To do this, go to Sketch -> Add Libraries, then select
your proper Zips.  This fixed a buggy UI for me.
Maybe it's a new Arduino thing, or could be a compatibility issue with the
OS I was using.