ramapcsx2/gbs-control #297: with l2c oled connected gbs board not responding

rellikami, 2021-12-07:
It happens only with d1 connected to the display all other screen connections don't cause the issue. Display works fine when fully connected and can navigate menus with the roatory encoder switch but doesn't do anything, browser navigation gives the same error with the screen connected but as soon as I disconnect d1 and do a reset it operated normally through the browser interface. Any idea what could cause this?

TABYDACAT, 2021-12-10:
Not sure how you have everything wired, but it possibly sounds like a wiring issue or even a power issue. I2C shouldn't have an issue in communication with only two devices. What's this the error that the browser is giving you when you have the OLED connected?

rellikami, 2021-12-13:
> Not sure how you have everything wired, but it possibly sounds like a wiring issue or even a power issue. I2C shouldn't have an issue in communication with only two devices. What's this the error that the browser is giving you when you have the OLED connected?
thanks for the reply it has been a long , but turned out to be a little bit of both, i got a beefier power supply (3amp 5v) that came in today i order for this project to power the gbs-control and a 5v audio amp curcuit. I am using the gbs-c to run an nesrgb to an 15khz arcade chassis board to fix a old broken sharp nintendo television. I tried shorter wires on the i2c display and it didn't initially do anything on the old psu, but when i wired in the new one it started work fine. i double checked the longer wires i had to be sure and got the old issue back.  anyway long story short TLDR everything is golden now, just have to figure out how to fit everything inside the tv shell soundly,

closed 2021-12-13 by rellikami