ramapcsx2/gbs-control #299: Csync issue

BuddiScotch, 2021-12-23:
Hi,

I am currently using a GBS-8220 with a external clock generator installed, 

I also recently modded my N64 with RGB output and am using a cheap EU scart cable to connect it to the GBS and encountered a sync issue where the displayed image is scattered and eventually looses signal entirely.

[image]

after looking through other github issues(#216, #113, and #161) on similar problems and a RetroRGB video, I decided trying 470, 330 and 300 ohm resistors each to pin 20 inside the scart cable but still got the exact same results 

[message.txt](https://github.com/ramapcsx2/gbs-control/files/7766961/message.txt)

this is the debug info from the console,
something to note is that at the very start on rare occasions it actually holds sync for around 5-10 seconds before it jumps to "no signal" and "source format not recognized" while flickering how you see above.

if anyone has suggestions or has a similar working setup with the N64 RGB mod, I'd appreciate any advice or feeback.

Thanks,

ramapcsx2, 2021-12-23:
Can you resolder your N64 mod to use sync on luma? That should work better.
Csync often requires specific attenuation (resistors of the right sizes). Luma sync is less tricky.

BuddiScotch, 2021-12-24:
sorry for the late response I was at work. I've removed the csync wire from the N64 and tested it and the image is much more stable for the minute or two I had it on however it still had some visual artifacts every other second that looked like a sync issue.

The scart cable I'm using still has a 470 ohm resistor still soldered to the sync pin and I'm assuming this is part of it, should I remove the resistor entirely or solder on a lower rating?

thanks again for responding,

BuddiScotch, 2021-12-24:
[image]

this is the image I get with out any resistor in the scart cable

BuddiScotch, 2021-12-26:
update:

after rewiring the cable to use sync on luma, I was able to get a stable image. Which on start up remains stable but only for a few minutes 

then it looses sync again

[image]

[image]

this time with different artifacts like the colours suddenly dimming, but it always gets scrambled by the end

ramapcsx2, 2021-12-26:
I think there's some kind of problem, possibly on the GBS board itself.
Do other consoles work fine?

BuddiScotch, 2021-12-26:
I was actually about to write an update about this today

after testing other consoles I think the worst has happened and I must've fried the board.

on all other consoles I now get a blank screen, sometimes I'll get some visual artifacts like this for the wii. keep in mind I was unplugging and plugging things back in at the time this photo was taken, and only the red rca jack was plugged in at the time
[image]

It worked perfectly fine with other consoles before, however when I opened up the NTSC scart cable (the one that I had moved the sync wire to pin 7 on the multi out end of the cable, which according to the chart below, was the s video pin) I had found that the wire going to pin 10 on the multi out end (shown in the second diagram below) was loose inside the cable.
[image]
[image]

I must've accidentally hit it with my soldering iron and it desoldered from the pin, which clearly shorted something and the GBS is no longer functional.

for every console that I try, in the console what I get is "Source format not recognized! using fallback preset" 

so, in the end it was my own fault for choosing a cheap cable and trying to do it myself, but thanks a ton for helping me out and being patient anyway

ramapcsx2, 2021-12-27:
Hm, tough luck there, but a learning experience for sure :)
Let me recommend the HD Retrovision Genesis cable with the various console adapters. This is ideal in quality and gives you Component Video for the GBS, which is much easier to deal with.