ramapcsx2/gbs-control #300: Power GBS through ESP

wasbaun, 2021-12-26:
Hi there, my 5v barrel power plug kinda died today and I had no replacement. So I tried using a usb charger and plugged it directly in the esp. Now the the GBS board is powered through the esp and it works. I watched the whole fellowship of the ring with now problem with my ps2 via component. The wirering itself was made after voultars guide, with multiple grounds.

My question is, if this could harm the hardware? I like the idea, powering the whole gbs-controll with a standard usb cabel instead of changing the plug on the GBS board.

ramapcsx2, 2021-12-26:
It is certainly operating the 3.3V regulator on the ESP board out of specs. This might work fine for a while, then fail suddenly.
I would repair the barrel connector :)

atinocoh, 2022-02-21:
I can confirm what @ramapcsx2 says, I burnt my 3.3V regulator on the ESP doing this haha

NO-ob, 2022-05-14:
https://www.adafruit.com/product/4090 i've been powering mine with one of these if you want an alternative to the barrel jack