ramapcsx2/gbs-control #304: EDTV preset distortion if Line filter is ON - possible solution?

lovrenco, 2022-01-01:
I managed to make Multiscan screen mode work using GBS control on my Amiga. Tried it on A1200, but will do on A600 too.
It took some screen positioning, but it looks fine now (I do not know why, but EDTV mode positions image too much to the left and up on 1280x1024 resolution?).

Multiscan Productivity mode on the Amiga is 640*480, max colors 256, refresh 60Hz, 31,44kHz
It is recognized as EDTV, and I needed to switch of "Matched presets" in order to get it to work with GBS1280x1024 resolution.

GBS switches to Pal Hig Res instantly (50Hz, 15,60Hz) what is needed when you start classic Amiga game from the menu shown on Image1. Great!
GBS Save slot supports different positioning/settings for different resolutions which is FANTASTIC JOB!!!!
That means I was able to tweak both options per my need and save them to the same SLOT (modes: preset_ntsc_480p.G and preset_pal.G). 

Multiscan Productivity is great because it is not interlaced so there are no mouse cursor artefacts on the screen as in Pal High Res Laced mode (I wrote about that before). 
Multiscan mode is supported by ECS and AGA Amigas, which means A500+, A600, A1200, A3000, A4000    

However, I noticed that "Line filter" option is not saved per preset, if you save it it is applied to both modes. Which leads to this problem I am trying to describe here. 

Image1: You can see my Workbench screen (multiscan mode), no line filter turned on (preset_ntsc_480p.G)
Image2: You can see my Workbench screen (multiscan mode), line filter turned on, notice garbage on the left side of the screen (preset_ntsc_480p.G)
Image3: PAL High res screen (game), line filter off. Notice horizontal lines, I marked them with arrows (preset_pal.G)
Image3: PAL High res screen (game), line filter ON. There are no horizontal lines, as expected (preset_pal.G) 

P.S. Line filter is only important if scanlines are turned ON.

**I thing easiest solution to this would be that "line filter" setting is saved "per resolution/preset" not globally. That way you could save it for screen mode they are needed for (since SAVE SLOT supports different settings for different mode - which is GREAT)!**

**_Console output (follows status of changing settings from Image1 to Image4_**
`.
2345678
Format change: 3 <stable>
preferencesv2.txt opened
loading from preset slot G: /preset_ntsc_480p.G
ADC offset: R:43 G:48 B:42
clock gen reset: 162000000
HTotal Adjust (skipped)

preset applied: 1280x1024 (custom) for EDTV 60Hz

source Hz: 59.15038 new out: 59.15038 clock: 161998096 (-1904)
Line Filter: off

2345678
Format change: 2 <stable>
preferencesv2.txt opened
loading from preset slot G: /preset_pal.G
ADC offset: R:43 G:48 B:42
clock gen reset: 108000000
HTotal Adjust (skipped)

preset applied: 1280x1024 (custom) for PAL 50Hz 

source Hz: 49.92035 new out: 49.92035 clock: 107958752 (-41248)
Line Filter: on
Line Filter: off

.
2345678
Format change: 3 <stable>
preferencesv2.txt opened
loading from preset slot G: /preset_ntsc_480p.G
ADC offset: R:43 G:48 B:42
clock gen reset: 162000000
HTotal Adjust (skipped)

preset applied: 1280x1024 (custom) for EDTV 60Hz

source Hz: 59.15038 new out: 59.15042 clock: 161998096 (-1904)`

Photos: 
[image]
[image]
[image]
[image]

lovrenco, 2022-01-06:
@ramapcsx2 Robert, have you seen this? Do you have any comment? I tried to give as much info as I could. 
Someone told me that you quit developing this great piece of software/mod, is that true?

ramapcsx2, 2022-01-06:
Sorry, can't help here :/

lovrenco, 2022-01-07:
@ramapcsx2 
> Sorry, can't help here :/

Not even this?

**I thing easiest solution to this would be that "line filter" setting is saved "per resolution/preset" not globally.**