ramapcsx2/gbs-control #306: Unable to change resolution

bobdillon, 2022-01-22:
As described [here](https://github.com/ramapcsx2/gbs-control/issues/246) by bambooshadow, I am encountering issues with the user interface.

Neither my iPhone nor my computer can select any options for resolution changes. 

Any resolution button I click seems to have no effect and everything is stuck on 15khz passthrough.

I can't get anything changed in the presets either.

Has anybody else encountered unresponsiveness in the web UI?  What is the fix?

zombizF, 2022-02-19:
Same issue... can't change the resolution, so i'm stuck the default resolution (1280x960)....

Odasan, 2022-09-13:
Hey

had the same problem, had weird problem in the video output, worked but couldn't change any settings in the gbs-c app, sometimes color were bugging etc

So what worked for me is relocate the clock power from C41 to C48 
Now everything works perfectly ! (At least for me)

Hopes that help