ramapcsx2/gbs-control #307: image is white and snowy

mastav79, 2022-01-22:
i used the build guide from voultars video along with the component from his amazon account. i am inputting standard AV and outputting using the vga to hdmi adapter in the component list on his amazon account. the image outputting is is colorless/white and snowy. since i followed the guide verbatim, i have no way to troubleshoot. the 5v to the vga to to hdmi adapter had white and red, so i used red for positive when i spliced the usb power to the breakout connector. i test all caps and resisters to verify they were the correct values recommended in his video. thanks in advance.

mastav79, 2022-01-22:
https://imgur.com/J1POYBe
https://imgur.com/94gJ40M

mastav79, 2022-01-23:
https://i.imgur.com/k2bf42k.jpg?1

ramapcsx2, 2022-01-23:
That is a Composite Video signal (yellow plug AV cable), right?
The board doesn't have a hardware decoder for encoded signals. It supports RGB or Component only.

ramapcsx2, 2022-01-23:
Ie: Your console has to be hooked up with either a SCART RGB cable (thick multi pin connector), or Component Video cable (3 + 2 plugs)

mastav79, 2022-01-23:
goddang it. lol. sorry im very new to scalers, scart, etc. whats the best solution to get an AV signal into a set of component cables?

closed 2022-01-23 by mastav79

ramapcsx2, 2022-01-23:
Instead of me trying to explain, here's the whole thing, but fun! :p
https://www.youtube.com/watch?v=VtTM7nU9SMA&list=PLTNBVisVMbSR1ZDDQRgjg6S9D2YQ4rwnZ