ramapcsx2/gbs-control #309: Black screen issue after it was working.

sincere360, 2022-02-04:
Hello everyone,

I built my gbs-control about a month ago and it's been working fine. I went to go play a game the other day and I get no picture, but I see the ESP8266 light up when the console is turned on. I tried different consoles and the same thing happens I see the blue light come that it's getting an input on the ESP8266, but the screen is black. To make sure it was not anything with SCART or the VGA to HDMI convertor I used VGA in to VGA out on a monitor.  My dreamcast has a vga cable so I connected that to the gbs-control and hooked up a monitor. I can see the monitor go from sleep to power on like it's getting a signal, but the screen stays black as well. Does anyone have any suggestions to where I can start troubleshooting this issue? I also re-flashed the ESP8266 and got the same results.

sincere360, 2022-02-05:
I figured out the issue. I saw someone in the shmups board had an issue with the purple hiletgo clock generators. I decided to disconnect it and see if I get a picture which I did. So I guess my clock generator went bad. I'm ordering an adafruit clock generator instead of another hiletgo purple to see if I have better luck.

K470r, 2022-02-16:
> 
Hello; you changed the clock generator and the problem is solved?

sincere360, 2022-02-16:
Yes, changing the clock generator resolved the issue. There is an issue with the adafruit clock generators. They cause a horizontal shakiness. I returned it thinking I had a bad part, but the replacement had the same issue. I ended up returning the adafruit and getting another purple clock generator. I'm not sure what is different in the adafruit part because I didn't compare datasheets.

K470r, 2022-02-16:
The problem I have is similar; GBS-Control worked for a while. One day I was playing and I started seeing a lot of lines of interference; then I turned it off and when I turned it back on it was on a black screen but the monitor detects that the video is connected. 

I'm going to change the clock generator, that must be the problem.

sincere360, 2022-02-16:
If you don't have a spare one on hand you can disconnect the clock generator and test. I played my consoles without issues while I was waiting for the new one. The higher resolutions did not work.