ramapcsx2/gbs-control #313: Tv says no signal with pal ps2 plugged into gbs control

Idontlikelongusernames, 2022-02-23:
I just want to start by saying I am by no means whatsoever very good with this kind of stuff. I followed a guide on YouTube for majority of it and double checked most of what I did on the wiki. To my point though I have no signal on my TV when plugging my ps2 into the gbsc. I can access the Web interface and change thing and I've changed a list of setting and also reverted everything back to the default on the Web interface however I've got nothing, I'm pretty sure my wiring is correct I could be wrong though. I just installed a new click generator hoping it would help. It was working without the clock generated however it was displaying a grey picture. I'm really not sure what it is. Help would be much appreciated

JAOT, 2022-07-27:
How are you connecting the PS2? Component? RGB? Composite? Scart with the Red, White and Yellow RCA Connectors?

Did you try the GBS board without GBSControl?

Saf98, 2023-04-25:
I have the same issue, getting weird artifacts.
I am using RGB Scart Sync on Luma from retrogamingcables uk. Same problem with RGBPi (Raspberry Pi 3) Cable.
I do not have the 100ohm resistor not sure if it's needed. Log shows no signal but does show rgb selected (oled display) with hz values fluctuating.