ramapcsx2/gbs-control #317: Source format wasn't recoginzed

PavelHardtkt, 2022-03-26:
Good morning to everyone, i'm into the Gbs-C from a while, and i absolutely love It.  I finish my second Gbs-C today, but when i plugged my PS2 with component cable It doesn't display anything. I can't select anything from the gbs local menu. I have this.

debug view: on
Source format not properly recognized, using fallback preset!
ADC offset: R:0 G:0 B:0
Active FrameTime Lock enabled, disable if display unstable or stays blank! Method: 0 (vtotal + VSST)
Warning: Check VDS_VS_ST!

preset applied: 1280x960 for EDTV 60Hz

I tried to swip the SDA and SCL wire, but It doesn't work. 

It can be a solder or wiring problem? Thanks

NathanJFV, 2022-04-09:
Same problem here.

tomudo, 2022-04-10:
I think gbs-control should have a debug wiki.

For general debugging.

1. GBS board - Connect gbs board without esp8266 for checking board is working.
2. Power supply - Check the power supply is 5v 2a or at least 5v 1a. Test with different adapter. 
3. ESP8266 - Check ESP8266 board still good (may be with blink example in Arduino ide, better with pins checking program and multimeter). 
4. Pin connection - Check pins connection are in correct order. 
5. Wires - Check wires condition with multimeter.

peon374, 2023-10-19:
> I think gbs-control should have a debug wiki.
> 
> For general debugging.
> 
> 1. GBS board - Connect gbs board without esp8266 and remove debug jumper for checking board is working.
> 2. Power supply - Check the power supply is 5v 2a or at least 5v 1a. Test with different adapter.
> 3. ESP8266 - Check ESP8266 board still good (may be with blink example in Arduino ide, better with pins checking program and multimeter).
> 4. Pin connection - Check pins connection are in correct order.
> 5. Wires - Check wires condition with multimeter.

Once you have done all that and still can't get 15KHz working what do you do next?  Is there a way to turn off automatically dropping 15KHz mode to pass-through so I can do other debugging?