ramapcsx2/gbs-control #320: Unstable IBM CGA via Microbee input board

pierre32-au, 2022-04-02:
Hi,

This is a bit of a story, and I hope I can get some insight here. I'm running a PC CGA card, and picked up one of these scalers to output to a VGA CRT:

http://www.microbeetechnology.com.au/store/cga-to-vga-converter-assembled-tested.html

Inside is a stock GBS 8200, paired with a custom input board to convert the CGA signal. You can see both boards in the photos, and there is a schematic for the Microbee input board here: https://i.imgur.com/YYzRCMH.jpg

Using this in stock form, the image is stable and the overall output fine. But there is room for improvement of the colours and sharpness, so this week I had the GBS Control update done to my board. There was an instant improvement to overall image quality, but now I have stability issues.

Here is a quick vid of the problem. Ignore colour & focus issues - that's my camera. It's just the snow/glitch effect which repeats constantly: https://www.youtube.com/watch?v=6sXlPjLf7PI

I have tested the GBS in isolation with a VGA input signal, and I can see that there are no issues with it. It would seem to be a quirk of the signal from the CGA card or Microbee board, only evident after the GBS Control update was carried out.

Any idea what might be causing it, and if it can be addressed in GBS Control options?

Log file attached.

[cga log.txt](https://github.com/ramapcsx2/gbs-control/files/8401464/cga.log.txt)

pierre32-au, 2022-04-02:
Reading through other issues, I saw @toms001 mention in #111 that disabling the Sync Watcher stops the issue. Indeed that works for me too. If this setting could be saved in a preset, I would consider it a permanent solution. But it's also no big drama to do it manually.

I also see in #295 that @Matty72 has the same Microbee unit I do. Not helpful to this issue, but nice to know others are using them for this.

"Print Info" log attached.

[PrintInfo log.txt](https://github.com/ramapcsx2/gbs-control/files/8401819/PrintInfo.log.txt)

ramapcsx2, 2022-04-02:
Hey,
yours is a very clean log, where I don't see any issues. Your video clearly shows sync losses though.

pierre32-au, 2022-04-02:
Hmm, wonder if I captured that output with Sync Watcher off. Here's some more, confirming this time that Sync Watcher is on. Apologies it's not text, but it's much easier to snipe with the print screen button!

https://imgur.com/a/6UhzhK2

ramapcsx2, 2022-04-02:
Yea, it's permanently unstable in these logs, while your video shows it would work otherwise.
This is a failure of one of the sync parameters, like sync windows, etc.
I can only ask you to try some earlier gbs-control versions, as these parameters got tweaked a lot.

pierre32-au, 2022-04-02:
Cheers :) I saw you'd suggested a 2019 version in #111 so I'll try that soon.

Thanks for all of your work.

pierre32-au, 2022-04-03:
Interesting results from the downgrade.

I want to be clear that I'm not asking for help here, as I accept the situation with my hardware. Just providing some notes.

With the old version linked in #111 I saw a new issue: The display will freeze completely as if the computer has hung, then resume motion. This repeats on roughly a 10 second cycle. The solution once again is to disable Sync Watch!

I could go down the rabbit hole trying other versions, but I reckon I'll just revert to the latest master. I know I can control the issue either way, and I get to enjoy the new web UI.

antaenc, 2022-05-25:
@pierre32-au I am working on a non-console RGB CRT to modern screen conversion. The video source has similarities to retro arcade machines. I've also found disabling the sync watcher gives a much improved stable picture. With it on I get a screen jump/flicker randomly but see nothing in the debug output. Had been chasing it down in the signals and HW I've built for ages! 

I don't fully understand the full GBS Control code, I was scanning through it this weekend; I wanted to disable sync watcher too without manual intervention. I've made a small addition to my local gbs-control.ino for the project. After this line https://github.com/ramapcsx2/gbs-control/blob/f45d25f3e2d4e564e47e29f00b49f5c527f48be7/gbs-control.ino#L3977 I set sync watcher to off: 

```
  }
  if (rto->videoStandardInput == 14) {
    SerialM.print(F("\nNote: scaling RGB is still in development"));
  }
  // presetPreference = 2 may fail to load (missing) preset file and arrive here with defaults
  SerialM.println("\n");

  SerialM.print(F("\nSyncWatcher at startup: "));
  SerialM.print(rto->syncWatcherEnabled);
  rto->syncWatcherEnabled = false;
  SerialM.print(F("\nSyncWatcher now: "));
  SerialM.print(rto->syncWatcherEnabled);
  SerialM.println("\n");
}
```

I'm still testing this and trying to iron out one very sporadic and random brief jump/flicker but this has improved my project no end for now.