ramapcsx2/gbs-control #321: How I can deactivate SyncWatch permanently?

NeoRame, 2022-04-15:
Since a few days I own a GBSC and try to use it with my NeoGeo MV1FZ

I found out that it works for me when I deactivate the SyncWatch BUT I can't save the change. After a reboot, sadly SyncWatch is activated again. Is there a way to deactivate SW permanently?

Sereosh, 2022-05-07:
Issues with MVS slots have been known for a while now. I tried disabling SyncWatch and indeed it appears to fix the jittery and freezing image! Hopefully a more definitive solution can be found in the future. [A user on AP reported an older version doesn't have this issue](https://www.arcade-projects.com/threads/mv1-fs-video-jittery.12967/post-328362) but I'd rather upgrade that go back to a version from 2019.

antaenc, 2022-05-25:
@NeoRame @Sereosh I too needed this, see: https://github.com/ramapcsx2/gbs-control/issues/320#issuecomment-1137196156

Snocksman, 2022-09-24:
I have the same Issue with SyncWatch on my Amstrad CPC 464. If SyncWatch is enabled, the Picture jumps from time to time and in special Display resolutions it only scolls diagonal over the screen for a while until the picture breaks down entirely.
If SyncWatch is disabled, all is fine. It would be really nice if I could disable SyncWatch permanently.

Snocksman, 2022-09-25:
I think I found a better workaround... Let SyncWatcher do his thing, and after that, disable it.
So I simply added **rto->syncWatcherEnabled = 0;** and now I have a really good and stable picture !

`    if (result == 0) {
        SerialM.println(F("Source format not properly recognized, using fallback preset!"));
        result = 3;                   // in case of success: override to 480p60
        GBS::ADC_INPUT_SEL::write(1); // RGB
        delay(100);
        if (GBS::STATUS_SYNC_PROC_HSACT::read() == 1) {
            rto->inputIsYpBpR = 0;
            rto->syncWatcherEnabled = 1;
            if (GBS::STATUS_SYNC_PROC_VSACT::read() == 0) {
                rto->syncTypeCsync = 1;
            } else {
                rto->syncTypeCsync = 0;
            }
            rto->syncWatcherEnabled = 0;   // All what SyncWatcher should do ist done now, so we can disable it       
        } else {
            GBS::ADC_INPUT_SEL::write(0); // YPbPr
            delay(100);
`

Snocksman, 2022-09-25:
I found a sollution which is working for me. I disable the Syncwatcher, after GBS Control has found and tuned to a stable picture once.
                   
 if (ignoreLength != GBS::SP_H_PULSE_IGNOR::read()) {
                        GBS::SP_H_PULSE_IGNOR::write(ignoreLength);
                        rto->coastPositionIsSet = 0; // mustn't be skipped, needed when input changes dotclock / Hz
                        SerialM.print(F(" (debug) ign. length: 0x"));
                        SerialM.println(ignoreLength, HEX);
                        **rto->syncWatcherEnabled = false;  // Disable SyncWatcher, after it has done it´s work...
                        SerialM.println(F("disabled syncwatcher"));**
                    }

crabfest, 2023-06-23:
> I found a sollution which is working for me. I disable the Syncwatcher, after GBS Control has found and tuned to a stable picture once.
>
> if (ignoreLength != GBS::SP_H_PULSE_IGNOR::read()) { [...]

Does anyone know how to apply this? I built 2 gbs, all components bought from different stores and both of them have issues, there's some weird pinkish noise that scrolls vertically in games that switch resolutions, they go away once syncwatcher is disabled.

Snocksman, 2023-06-25:
Applying this to your GBS Control is very simple.

Open the Code in Arduino IDE and search for the Code "block" mentioned in the post (searching for "if (ignoreLength !=" should be enough to find the right Part of the code. Then simply compare the code you see in the IDE with the one you see in the post and add the missing piece of Code (in the IDE).
Then compile and Upload.
What I changed is the following: When you Power up your GBS Control, Syncwatcher starts and runs until it finds a Signal and synced to it. After that Syncwatcher is disabled until next start of GBS Control.