ramapcsx2/gbs-control #326: Different issues with 2 out of 3 arcade boards.

rfancella, 2022-05-07:
I have used the exact same board on three arcade boards. I am using an ESP8266 with an Si5351 on board. The board works great on one arcade board and my Amigas but two arcade boards are not.
One is on a Raw Thrills Superbikes. The screen rolls. The other is a Harley Davidson & LA Riders Sega Model 3 Step 2 board. This works, kinda. The whites/greys are washed out.
What additional info can I provide?

[image]

[image]

Thank you,
Ron Fancella

tomudo, 2022-05-08:
Do you try reduce ADC gain?

f4f4f4f4, 2022-07-25:
I just worked on a Sega Gunblade, made me wish I hadn't replaced the pots with jumpers. Sega Gunblade and Midway/Nintendo Cruis'n Exotica have very high levels on the RGB lines, need some kind of resistor (or pot!) to turn those levels down. No setting in GBS-Control software gets the levels low enough.