ramapcsx2/gbs-control #332: Garbage on screen during resolution changes

JaCkBoston, 2022-05-24:
The setup: a PS2 Slim 90004, with a component cable running into the GBS, then through a lag-less VGA-HDMI adapter into the display.

The issue: When changing from interlaced to progressive scan modes the picture fills with garbage for a second, loses sync, blackouts for another, and then comes back to normal. The same happens between NTSC/PAL output changes.

How to replicate: Switch resolution on the fly while the GBS is running.

The issue can be seen in my video here: https://youtu.be/7csHB8WxTgk

This doesn't seem to happen in other people's setups and most people show the GBS to switch resolutions almost instantly and effortlessly. 

I must add that this doesn't seem to happen with SCART cable when changing from PAL to NTSC and back, but I can't check 480p as it doesn't work due to sync-on-green not being supported, I guess.

Is this an issue with the GBS itself or the cables/PS2?

Finally, here's a picture of my unit(currently without enclosure as I wait for the model to get printed). Could the longer cable to the external clock generator cause something like this? I don't experience any sync problems apart from this.

[image]

ramapcsx2, 2022-05-24:
Those are pretty normal for these kind of switches (between standard res and the "HD" modes).

JaCkBoston, 2022-05-24:
> Those are pretty normal for these kind of switches (between standard res and the "HD" modes).

Hello rama, so it's nothing out of the ordinary, and changing caps for sync-on-green as you suggested in this issue (https://github.com/ramapcsx2/gbs-control/issues/219) won't mitigate this behaviour?

ramapcsx2, 2022-05-24:
Internally, the jump from ~240 active lines to ~480 (or more) is a big change. It causes a full reconfiguration of the chip, which is what causes the garbage and also a lot of sync issues.
I tried a few tricks to mask this as best as possible, so what you get is already the improved version :p 

Better parts (caps) won't be able to fix this, as the problem is that full reconfiguration, which is definitely necessary.

ramapcsx2, 2022-05-24:
In fact, only a few select resolution changes are able to do with just minor reconfigs: Interlaced to progressive (480i to 240p) or the other way around, for example.

JaCkBoston, 2022-05-24:
I see, thanks for the clarification, Robert. I'm closing this since it's a non issue :)

closed 2022-05-24 by JaCkBoston