ramapcsx2/gbs-control #335: SSD1306Wire.h: No such file or directory

summerof19691969, 2022-06-11:
I watched a video and looked at the guide here and over and over for how to install the software to the arduino and am getting an error with this

Arduino: 1.8.19 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:1MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

gbs-control:23:25: fatal error: SSD1306Wire.h: No such file or directory

 #include "SSD1306Wire.h"

                         ^

compilation terminated.

exit status 1

SSD1306Wire.h: No such file or directory

Invalid library found in E:\Arduino\libraries\ESPAsyncTCP-master: no headers files (.h) found in E:\Arduino\libraries\ESPAsyncTCP-master

Invalid library found in E:\Arduino\libraries\ESPAsyncWebServer-master: no headers files (.h) found in E:\Arduino\libraries\ESPAsyncWebServer-master

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

What do I do?

mo-lab, 2022-07-11:
go to https://github.com/ThingPulse/esp8266-oled-ssd1306
download the zip file and add it to your IDE manually 
by going to sketch -> include library -> add .zip library

mkubiak402, 2022-11-19:
i did that still the same error..

vitecd, 2022-11-20:
Isn't it the Arduino library?

mo-lab, 2022-12-07:
replace it with:
#include <Adafruit_SSD1306.h>

it might work
then if you are still having problems with your code I suggest using PlatformIO instead of Arduino IDE
[https://www.youtube.com/watch?v=5edPOlQQKmo](url)

HornheaDD, 2023-03-24:
I am having this same exact issue. I followed the RetroRGB tutorial as well as tried to follow the Software Setup guide and I am still getting this error.

Ive got 3 COM ports that show up and all 3 of them end up with the same exact error. I have installed the libraries, I tried compiling, I tried downloading gbs-control-master, extracting and removing -master, and nothing is working. 

Ive tried it on 2 ESP boards and the same thing is happening.  What am I doing wrong here??

acjohnson, 2024-10-18:
It's too bad no one ever took the time to respond to this until now, but I ran into the same thing after watching Voultar's GBS-Control video. The problem is that I put the libraries in the incorrect location...

What helped me was to just use the Arduino IDE UI to install the libraries from the downloaded zip archives (eg. Sketch -> Include Library -> Add .ZIP Library)

In this case I just browsed to the `esp8266-oled-ssd1306-master.zip` file and it installed into the correct location instantly solving this error message.

And I don't think this is a misstep in Voultar's video, it's just that I screwed up the steps myself :stuck_out_tongue_winking_eye:

sirjeannot, 2024-11-16:
installing this library from the package manager solves the dependancy issue:

[image]

no need to manually install it using the zip.
the issue is misleading as one would expect the expected dependancy is "SSD1306wire", but no.