ramapcsx2/gbs-control #337: Preset Select with Hardware Switch

ILAHWWINC, 2022-06-12:
3 years ago rama send me a code fragment to pit in the sourcecode. 
The result was the possibility to cgange between Preset 1 and Preset 2 with a hardware switch. 
Will it be possible to do the same with the actual code?

https://github.com/ramapcsx2/gbs-control/issues/66

jsteel44, 2022-06-24:
I just found that old post and adapted it to work with the latest code. This works for me, I'm no expert but the preset gets applied only when I flick the switch so I think it's a good place for it, I only had to change preset 1 and 2 to A and B, otherwise it's identical to what we suggested before:

```
--- a/gbs-control.ino
+++ b/gbs-control.ino
@@ -6789,6 +6789,29 @@ void runSyncWatcher()
     SerialM.println();
     goLowPowerWithInputDetection(); // does not further nest, so it can be called here // sets reset parameters
   }
+
+  static const int switchPin = 13;
+  pinMode(switchPin, INPUT_PULLUP);
+  static uint8_t oldSwitchState = 1;
+  int switchState = 1;
+  if (detectedVideoMode > 0) {
+    switchState = digitalRead(switchPin);
+
+    if (switchState != oldSwitchState) {
+      //uopt->presetPreference = 5;
+      if (switchState == 1) {
+        SerialM.println("slot A");
+        uopt->presetSlot = 'A';
+      }
+      else if (switchState == 0) {
+        SerialM.println("slot B");
+        uopt->presetSlot = 'B';
+      }
+      saveUserPrefs();
+      applyPresets(detectedVideoMode);
+    }
+    oldSwitchState = switchState;
+  }
 }
 
 boolean checkBoardPower()
```

Just apply with `$ patch -N -i switch.patch`. I thought I would just have to wire to 3.3v on one side of the switch, but it wasn't working (nothing seemed to happen at all) until I wired the other side to ground, so now the switch toggles D7 between 3.3v and ground and it works as expected.

israpps, 2022-09-07:
what about the `handleButtons()` and all the code stored inside `HAVE_BUTTONS` macro?

When i get more confident on this one i Might make a modification to make the down and up buttons traverse the preset array.

israpps, 2022-09-09:
I made this crap
https://github.com/israpps/gbs-control/tree/preset-traverse-button
Any of you thinks it will work??

I've seen the oled code uses chars from `A` to `G` to pick presets.

Not sure if i should do the same