ramapcsx2/gbs-control #340: Image suddenly pink for all sources? Is she cooked?

tunnotron3000, 2022-06-22:
Hi all! I've been using my GBS without much issue for about a year now. I tried a different VGA to HDMI dongle yesterday and the picture was suddenly pink. I chalked it up to the converter itself and went back to chaining it into my OSSC but found the image was still pink.

Since then I've reflashed the firmware and inspected the soldering and everything seems fine.

Before I pull out the soldering iron to start disassembling things, does anyone have any idea what may have happened? Is it possible the dongle has fried the board somehow?

Weirdly enough, trying the component output gives a normal coloured but dark image (as expected).

Thank you!

ramapcsx2, 2022-06-22:
Sounds like a problem too difficult to remote debug :p

tunnotron3000, 2022-06-22:
> Sounds like a problem too difficult to remote debug :p

That's ok, I'll just grab a new board as they're fairly cheap - the beauty of the GBS project! Thanks for replying though Rama, appreciate it!

closed 2022-06-22 by tunnotron3000