ramapcsx2/gbs-control #341: RGsB (Sync on Green) from PS2 --> No Signal detected

nor2101, 2022-06-24:
Hi,

I've recently modded the GBS8200 with the gbs-control and it has been relatively smooth sailing.

In my brief testing so far all the inputs work fine (YPbPr, RGBHV, RGBs) except for the RGsB one (connected to the d-sub 15 port where you connect rgbhv vga input).

I'm testing the console with a secondary RGsB compatible VGA monitor and I'm sending the signal using the ps2 component cables + a simple passive 3 RCA to D-sub 15 adapter, and when I set the console in VGA 640x480 mode I get a picture just fine.

Sadly with the same setup but using the gbs-control I get no picture (no blue LED active at all). When I set the console to regular RGBs using the same passive adapter and I also connect the yellow wire of the ps2 component cables to sync (tip to sync and sleeve to gnd of course) I get a signal from the GBS8200 just fine.

Somehow the GBS can't find the sync signal on the green line.

I tried both with and without a 100Ohm resistor between sync (csync) and gnd: same result.

Now, I have a few doubts:
- Does the GBS actually support RGsB or do you need an external sync strip IC (like the LM1881)?
- Could I have damaged some traces when desoldering the pots rp1, rp2, rp3? I have continuity between the green line and the R56 75Ohm terminator resistor, but I'm not sure if the signal should also be going directly to C33 and C35 (the SOG capacitors).
I can't find any actual schematics of the board online so I don't know what to check if anything.
- Do I need to set any specific software setting or should the signal be detected normally like the others? I couldn't find anything related to SOG.

Thank you for your help and patience.

ramapcsx2, 2022-06-24:
I think the problem is that the RGBHV / RGBS input channel is not programmed to look for RGsB. Doing so had several problems, iirc.
It should work if you use the 3 input port.

nor2101, 2022-06-24:
> I think the problem is that the RGBHV / RGBS input channel is not programmed to look for RGsB. Doing so had several problems, iirc. It should work if you use the 3 input port.

Thanks! I tried with the 3 input component port but I get a strong red tint. The GBS sees the sync on the Y wire but still thinks it is component video and tries to process it like YPbPr.

I experimented and tried to connect the green line to C33 with a jumper --> I get a picture on rgbs input! But it has a slight rec colour cast. I removed the jumper immediately because I don't know if it could cause damage to the GBS.

ramapcsx2, 2022-06-24:
But I think I have a special mode for the PS2, that allows the console in RGBHV mode to run higher resolutions over the Component cables. I really don't remember well though..

nor2101, 2022-06-25:
> Yep, the Component input is meant for Component, not RGsB. The difference is in the color mixing.
> 
> You would not damage the GBS, I think, but yea, you found the "issues", or one of it :)

Ok, thank you for the explanation. Do you think that the reason for the slight red colour cast when bridging green to C33 is because somehow by doing so it's lowering the voltage of the green line? I know that if you split a video source without amplification you get a darker or no image at all (and in some cases you risk damaging the video source), but perhaps this is a different issue altogether.

> But I think I have a special mode for the PS2, that allows the console in RGBHV mode to run higher resolutions over the Component cables. I really don't remember well though..

It's ok. I'm building an RGB Scart breakout board anyway, might as well throw an LM1881 in there to extract sync from green, even thought I'm not really using it that much (it was curiosity more than anything, ahahah).

Thank you again for the answers (and for this incredible project!)

ramapcsx2, 2022-06-25:
You're welcome :)
Color balance problems come from the design differences between Component Video and RGsB. 
They have different electrical properties, and the decoding hardware and software has to be adapted to which it is.
The idea of bridging sync into the channel somehow won't work, I think.

But look into that special PS2 mode more. There should be a mode that allows you to run high RGB resolutions with correct color, and without special circuits.

nor2101, 2022-06-25:
> You're welcome :) Color balance problems come from the design differences between Component Video and RGsB. They have different electrical properties, and the decoding hardware and software has to [...]

I tried that: I can output DTV resolutions (480p, 720 and 1080i) from ps2 component video, but any VESA mode (640x480, 800x600, 1024x768) forces sync on green, even if you set YPbPr in the system configuration.

AFAIK there is no way to get RGBs 480p or higher on PS2 through software, only vesa 480p sync on green or DTV 480p component.

I know it's possible to hardware mod the ps2 to disable SoG and always force csync in RGB mode, but I couldn't find more info about it and I'm not comfortable poking around on the ps2 mobo without guidance.

ramapcsx2, 2022-06-25:
I think the special mode is that kind of PS2 sync on green still over Component. It should give you a picture that is correct, for the VESA modes.
https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L5052

nor2101, 2022-06-25:
> I think the special mode is that kind of PS2 sync on green still over Component. It should give you a picture that is correct, for the VESA modes. https://github.com/ramapcsx2/gbs-control/blob/master/gbs-control.ino#L5052

Ok.

I tried the other vga vesa modes though component and: they work!

The ONLY one that gives the pink tint is VESA 640x480@60Hz.

All the others are decoded properly, as you can see:
[image]
[image]
[image]
[image]

ramapcsx2, 2022-06-25:
Right, that's a cool hack that kind of works okay. Find a good mode for OPL and you can even run games with this :)

nor2101, 2022-06-25:
> Right, that's a cool hack that kind of works okay. Find a good mode for OPL and you can even run games with this :)

Yeah, it's pretty cool! Too bad for vesa 480 60Hz, but I guess it's not worth the hassle to make the software detect it differently from 480p component

ramapcsx2, 2022-06-25:
That's the main issue of this project, that causes most of the unreadable code: Too many possible variations, and they only have little differences to detect them. This meets too many possible operation modes, and it all became a mess :p

kamikaze-star, 2025-05-28:
no input ,no input ,no input pura basura