ramapcsx2/gbs-control #342: Override/Bypass Rgb Potentiometers

JAOT, 2022-06-30:
Hi people!
For the sake of brevity, let's correctly assume I made a bad job removing the potentiometers, as probably the soldering iron was way too hot. Now it's only used at the minimum temperature. 

So, blackened traces and I believe the bridging will not work, as the damage was severe, in my opinion. 

[image]

Can I make a connection directly from the Rgb pins in P11 to R55, R56, R57? I could use the underside of the board to do so.

Sorry for submitting something akin to tech support and not an issue. 

A schematic could help me, but I was not able to find if P3, P10, P11 share the same traces, but I believe they do. As I'm going for Scart, P11 is the obvious choice, and has a convenient access on the underside. 

Thanks.

ramapcsx2, 2022-06-30:
You can see where the traces are supposed to lead, and those connections should be bridged properly.
The damage has to be multimeter checked beforehand, and any shorting to ground or other traces has to be fixed.

JAOT, 2022-07-01:
Just a quick follow-up.
Made sure all was well, checking connections and went on with my guess:
[image]

After some crazy testing with jumper cables and some patience (and connecting a missing ground wire on SCART connector), I got a picture:
[image]

And then made the change permanent:
[image]

I am not sure if the image is the absolute best, as I hand-made the RGB cable with cables I had lying around. In the power bar and backdrop there's a bit of a mosaic, but I am not sure that is supposed to look like that. Same thing happens with the health bars in Street Fighter II.
I have the rest of the materials to further improve GBSControl, and will proceed asap (as in: when I finish my degree in 15 days!)

My soldering isn't the best, but I made it work!

Thanks @ramapcsx2 for making our childhood companions live a happy life in these modern times!

closed 2022-07-01 by JAOT