ramapcsx2/gbs-control #343: Missing Green throught SCAR RGB

Ged91, 2022-07-08:
Hi, I did the mods but it seems that green colour is not coming out. I'm using a pal snes through rgb scart cable, that is plugged into P11. Going to the TV is a VGA to HDMI adapter. I checked multiple times my wiring on the female SCART and everything seems ok to me. I'm afraid I may have damaged the GBS PCB while removing the pots. Can anyone suggest some good testing points to find out where green is interrupted? Also, I'm not getting continuity between RP2 and R56, is that normal?

https://imgur.com/a/U0jQ0A4

JAOT, 2022-07-08:
It's dangerous to go alone.
Try this: https://github.com/ramapcsx2/gbs-control/issues/342

Ged91, 2022-07-08:
> It's dangerous to go alone. Try this: #342

Thank you. That's I was planning to do but I figured I should check with you guys before. I ended just bridging RP2 and R56 and it's working now.

https://imgur.com/a/uzReAEo

JAOT, 2022-07-09:
You can close this one, no?

closed 2022-07-09 by ramapcsx2