ramapcsx2/gbs-control #344: Vertical lines/banding

Ged91, 2022-07-09:
Hi, I've completed my build and the image looks fine, but there is this faint vertical banding: https://imgur.com/a/IoDTHPD

I did the 4 parallel caps and also the 22uF in place of C11. Any idea of how can I fix this? Thanks

JAOT, 2022-07-09:
What is the connection?
Are the cables shielded?

Is the power supply any good?

ramapcsx2, 2022-07-09:
These look typical for bad cables, yep. Some consoles (and revisions) also have significantly more banding than others.

Ged91, 2022-07-10:
This is on a PAL SNES with scart RGB. Are guys talking about the scart cable of the wires I used for the mod? The scart cable is supposed to be good.

Can you guys point me to a known good PSU?

I'll try some other consoles

SteppoBlazer, 2022-07-10:
I had the exact same issue you had, persistent noise patterns in the image output of various consoles. Here are my two cents on how I managed to fix my issues, although be aware that I use the GBS-C on a Fujitsu CRT VGA monitor and not an LCD panel so not sure if it will make any difference in your case. My GBS-C has the Clock-gen module installed, a Euro-SCART RGB connector and a 680 ohms resistor on the video input line of the SCART Socket. I can confirm that a high quality PSU and adding 10uf / 22uf ceramic SMD capacitors in parallel to stock ones C23, C41(alternative: C43), C42, C48 helped a ton and played a big role in getting rid of much of the interference patterns I had. Tho even after all those mods there were still some very tiny interference patterns showing up occasionally, specially on images with single solid colors. So after trying with different capacitance caps, for my case scenario, using 100uf caps removed any form of interference from the image. Tested the GBS-C with Genesis, SFC, PS1, XBOX, PS2, PSP, XBOX 360 and WII. Genesis, SFC, PS1 outputting RGB via high quality double shielded RGB Cables and the rest Outputting Component using official PSP Component cable, official WII Component cable, official XBOX-360 Component cable and for the XBOX and PS2 I modified official XBOX-360 Components cables, so they should also provide high quality outputs. Hopefully my advice will help ya, cheers!

Ged91, 2022-07-10:
> I had the exact same issue you had, persistent noise patterns in the image output of various consoles. Here are my two cents on how I managed to fix my issues, although be aware that I use the GBS-C on a Fujitsu CRT VGA monitor and not an LCD panel so not sure if it will make any difference in your case. My GBS-C has the Clock-gen module installed, a Euro-SCART RGB connector and a 680 ohms resistor on the video input line of the SCART Socket. I can confirm that a high quality PSU and adding 10uf / 22uf ceramic SMD capacitors in parallel to stock ones C23, C41(alternative: C43), C42, C48 helped a ton and played a big role in getting rid of much of the interference patterns I had. Tho even after all those mods there were still some very tiny interference patterns showing up occasionally, specially on images with single solid colors. So after trying with different capacitance caps, for my case scenario, using 100uf caps removed any form of interference from the image. Tested the GBS-C with Genesis, SFC, PS1, XBOX, PS2, PSP, XBOX 360 and WII. Genesis, SFC, PS1 outputting RGB via high quality double shielded RGB cables and the rest outputting Component using official PSP Component cable, official WII Component cable, official XBOX-360 Component cable and for the XBOX and PS2 I modified official XBOX-360 Components cables to provide high quality outputs. Hopefully my advice will help ya, cheers!

> I had the exact same issue you had, persistent noise patterns in the image output of various consoles. Here are my two cents on how I managed to fix my issues, although be aware that I use the GBS-C on a Fujitsu CRT VGA monitor and not an LCD panel so not sure if it will make any difference in your case. My GBS-C has the Clock-gen module installed, a Euro-SCART RGB connector and a 680 ohms resistor on the video input line of the SCART Socket. I can confirm that a high quality PSU and adding 10uf / 22uf ceramic SMD capacitors in parallel to stock ones C23, C41(alternative: C43), C42, C48 helped a ton and played a big role in getting rid of much of the interference patterns I had. Tho even after all those mods there were still some very tiny interference patterns showing up occasionally, specially on images with single solid colors. So after trying with different capacitance caps, for my case scenario, using 100uf caps removed any form of interference from the image. Tested the GBS-C with Genesis, SFC, PS1, XBOX, PS2, PSP, XBOX 360 and WII. Genesis, SFC, PS1 outputting RGB via high quality double shielded RGB cables and the rest outputting Component using official PSP Component cable, official WII Component cable, official XBOX-360 Component cable and for the XBOX and PS2 I modified official XBOX-360 Components cables to provide high quality outputs. Hopefully my advice will help ya, cheers!

Thank you for the thorough answer. Where exactly did you use the 100uf caps? On C23, C41, C42 and C48, in place of the 10uf?

SteppoBlazer, 2022-07-10:
Yup. At first I tried with the adviced 10uf up to 22uf and they helped mitigating the interferences. Then I decided to up the capacitance to 47uf and the difference was very noticeable. So I had some 100uf 25v spares hanging around and decided to try those on to see what would happen and well... All noise was gone!

Ged91, 2022-07-10:
All right, I guess I'll give it a try. Thanks

gabrielgamer99, 2023-08-17:
> All right, I guess I'll give it a try. Thanks

Hello I was wondering if you ever tried the 100uF capacitors, currently having the vertical banding you had and was wondering if this is a good solution.

Ged91, 2023-08-17:
> > All right, I guess I'll give it a try. Thanks
> 
> Hello I was wondering if you ever tried the 100uF capacitors, currently having the vertical banding you had and was wondering if this is a good solution.

Didn't yet, life got in the way. But I was just about to try this days, I'll let you know.

Firas-Dhouibi, 2024-01-01:
i have the same issue, did u try the 100uf caps ?

Ged91, 2024-01-02:
> i have the same issue, did u try the 100uf caps ?

Didn't have the chance yet, but I plan to do it soon. I'll report back to you when I do.

Firas-Dhouibi, 2024-01-13:
ahh yes, i forget to ask, when u change the preset to let's say 1080p these lines will be less visible ?

cpawliuk, 2024-07-28:
> Yup. At first I tried with the adviced 10uf up to 22uf and they helped mitigating the interferences. Then I decided to up the capacitance to 47uf and the difference was very noticeable. So I had [...]

I'm waiting on a board to do this and thinking of just removing the SMD caps and putting some 100uf 16v, not sure if I need to go with 25v? Not really a fan of stacking 0805s in a tight space like that. Have you had any issues since you put 100uf 25v electrolytic capacitors in?

ramapcsx2, 2024-07-28:
Adding more caps if you already have some extras usually doesn't help any more.
Try using a different SDRAM clock (web ui developer option). This changes the noise characteristics the ASIC generates, and sometimes helps to hide vertical band noise in the output.
But.. if the banding is present in the source console, this won't do anything at all.

cpawliuk, 2024-07-28:
> Adding more caps if you already have some extras usually doesn't help any more. Try using a different SDRAM clock (web ui developer option). This changes the noise characteristics the ASIC [...]

This is a relief if I don't need to stack those SMD capacitors over the existing ones! Thanks again for all your work @ramapcsx2 

Just to confirm, the hardware mod: https://ramapcsx2.github.io/gbs-control/Wiki/Hardware-Mod-Library.html

"Add 10uf / 22uf ceramic SMD capacitors in parallel to stock ones (x4) C23, C41(alternative: C43), C42, C48"

Is no longer needed?

What about: "Remove C11, optionally replace with 22uF (6.3V to 16V) electrolytic cap" ?

Thank you!

cpawliuk, 2024-07-28:
> Adding more caps if you already have some extras usually doesn't help any more. Try using a different SDRAM clock (web ui developer option). This changes the noise characteristics the ASIC [...]

Thanks again for all your work @ramapcsx2 This project really shows a lot dedication!

Just to confirm, the hardware mod: https://ramapcsx2.github.io/gbs-control/Wiki/Hardware-Mod-Library.html

"Add 10uf / 22uf ceramic SMD capacitors in parallel to stock ones (x4) C23, C41(alternative: C43), C42, C48"

Is this no longer needed? Looking at the videos of this being done it seems to be such a tight spot and a bit of a pain to stack them so close near other components. It just seems that it would be easier to just remove those four and replace them with a electrolytic capacitor.

If this is still required, were you confirming that going from the stacked recommendation of those SMD capacitors to putting in 100uf 25v instead is not beneficial?

What about: "Remove C11, optionally replace with 22uF (6.3V to 16V) electrolytic cap" ?

Thank you!

ramapcsx2, 2024-07-28:
Some misunderstanding here :p

Adding the SMD capacitors as shown is beneficial, up to a certain number, each additional one helps.
It has to be SMD ceramics, as other types aren't "fast enough" to be useful against noise.

What I meant to say is that, if you already have added some capacitors, then adding any more usually won't help.

cpawliuk, 2024-07-28:
> Some misunderstanding here :p
>
> Adding the SMD capacitors as shown is beneficial, up to a certain number, each additional one helps. It has to be SMD ceramics, as other types aren't "fast enough" to [...]

No worries! I was debating to just remove the SMD caps there and just put a 100uf 16v/25v electrolytic capacitor in its place as I had come across a few posts where someone was doing this and said the visual interference was gone. So I thought to at least ask here before doing so.

I'll need to practice a bit on my test board stacking those SMD caps in such a tight space. Not something I've done before.

ramapcsx2, 2024-07-28:
In theory (and if you'd watch the noise on a scope), swapping SMD against electrolytes will make the noise worse.
In practice, this is a very weird field, and it might actually / accidentally work :p

cpawliuk, 2024-07-28:
> In theory (and if you'd watch the noise on a scope), swapping SMD against electrolytes will make the noise worse. In practice, this is a very weird field, and it might actually / accidentally work :p
> 
> (Why it might work: There will be more noise then, but the characteristics of it will change, which could make it so that the particular artefact in question might be masked... Expect others to show up then though, as generally, more noise = worse.)

Thank you for the clarification.

Kuglibugli, 2024-09-02:
Hello! I would have an additional question to this if you don’t mind!

So, it is said to solder the extra capacitors parallel to the original ones, however wouldn’t it be better to simply swap the original capacitors to the 10uF ones?

thank you very much!

ramapcsx2, 2024-09-02:
In my opinion, it is easier to just add them in parallel, but it could be different for you, dunno :p 

However, the way bypass caps work, it is advantageous to have several of them in a parallel configuration, if they each have a different (capacity) value.
This has to do with their ability to bypass (to ground) noise at different frequencies, where smaller cap values are more effective at higher frequencies, larger caps better at lower frequencies.
Ideally, there should be a mix of them that covers the entire range of noise, which is often 3 of them, at something like 1nF, 100nF and 1uF... for devices of this ASIC's vintage. More modern devices have lower values.

cpawliuk, 2024-09-02:
Hello @ramapcsx2 , do you by chance have a link or access to schematics for the GBS 8200 or do you have any notes on the values of the SMD caps throughout the board?

I did want to ask, do you prefer soldering the caps on the underside of the PCB as opposed to directly on top? The caps next to the IC are pretty crowded so I was considering that as an option. I did get my board so I'm looking forward to undertaking this project. Tad bit nervous on the stacking part due to such a tight space. Either way I'll be putting in some time on the practice board before I go ahead with this.

Thank you.

Kuglibugli, 2024-09-02:
> In my opinion, it is easier to just add them in parallel, but it could be different for you, dunno :p
>
> However, the way bypass caps work, it is advantageous to have several of them in a parallel [...]

That does make sense! I have the tools to replace them, would not be much of a worry. Do you possibly know the capacitance of the stock ones? Anyway I could remove one and check worse comes worst.

the reason why I am asking this because I have been dealing with this issue seen on this recording and could not figure out what causes it… does it with my PS2 and with a simple vga passthrough as well:
https://youtu.be/9Wn3tQ6hB9o?si=PzvRCvL5M-SeDmVv

ramapcsx2, 2024-09-02:
@Kuglibugli That is not bypass cap related, that's some issue possibly with a HDMI dongle or such.. Maybe clock gen / no clock gen.

@cpawliuk Nope, not really :p

Kuglibugli, 2024-09-03:
@ramapcsx2 Had a bad feeling yes. Thank you anyway! Will investigate.

cpawliuk, 2024-09-03:
> @cpawliuk Nope, not really :p

Yea, I didn't think it would be that fun to do the underside, especially since you need to remove some of the mask in the proper spots to get at it. Thinking about it now I'll just do the smart thing and use kapton tape to isolate the areas off and give myself a more safe approach. My main concern was just the other components being so tightly nested, and the IC right next to the caps.

I'm assuming you don't have the schematics as well? Seems to be impossible to find one. 👀

Thanks again! 👍

ramapcsx2, 2024-09-03:
Please don't overthink the caps thing. I had overall mixed results with this, with most of the benefits only being visible on an oscilloscope, watching the power rails or the RGB analog output when displaying low intensity primary colors etc.
The biggest source of banding etc seems to be the consoles themselves, and the GBS ASIC doesn't have the best input filter before its ADCs to fix them.
Improve the console noise (caps again :p) and it should have a bigger impact.

cpawliuk, 2024-09-03:
> Please don't overthink the caps thing. I had overall mixed results with this, with most of the benefits only being visible on an oscilloscope, watching the power rails or the RGB analog output when [...]

What I could just do is have all the other stuff done. ie. Clock Generator, remove or replace the cap on C11, remove the 3 RGB potentiometers, ect... and if needed look at adding more capacitance for power supply bypassing at later point in time.

I also got a LCR Meter on order and a second GBS 8200 board I'll look through later to try to grab some values and learn more about the board overall as I can remove certain components to test them. As much as I'm doing this to build a nice device for my retro consoles, I'm also extremely interested in the entire process.

Looking forward to getting the final build done soon as I got all the parts I need to finish it! Thank you @ramapcsx2 !