ramapcsx2/gbs-control #352: Please help with resistor values

ghost82uk, 2022-07-25:
Good evening all

I recently made the decision after a few chats to put together a GBS 8200 with the GBS Control board fitted. Followed the GitHub with the help of some videos and got it to a point where GBS Control loads, I can connect to it over wifi, but I couldn't get a stable sync using my GameCube to test. I found a guide that was talking about removing resistors R32 and R34, replacing R34 with a 75-ohm resistor instead of adding the 100-ohm resistor between sync and ground. I tried this with no luck. Can someone tell me the original values of the 2 resistors I removed as I have lost them so I can go back and try something else

Thanks all

f4f4f4f4, 2022-07-26:
I don't think R32 should be removed.
R32 and R34 are 1kΩ? (SMD "102")
I have a 100Ω resistor across the sync and ground, and measuring them in-circuit:
R32: 159Ω
R34: 83Ω (this is where we are close to the 75Ω target)

ghost82uk, 2022-07-26:
that's perfect, thanks for that, yeah 102 are 1kohm resistors, they must be in parallel as I found stuff talking about them having a 500-ohm resistance so that makes sense. Thank you so much for the help there. I'll get them soldered back in tomorrow as I should have 1k 0603's in stock