ramapcsx2/gbs-control #353: Losing wifi interface connectivity when in 1920x1080 mode

DubbyMcDubs, 2022-07-26:
Hi. Just received my GBS-C from aliexpress yesterday and have been getting great results so far with the PS1. Great job!

I have flashed it with the lastest code, however I am still experiencing an issue where I loosing connectivity to the control web interface when running in 1920x1080 mode. This means I have to change resolutions to change other settings. This is not great as I cant scale the output etc for that particular high res.

Any suggestions.

Thanks!

leon14344, 2022-07-29:
I'm having the same issue, but from Component from a PS2. The moment 1920x1080 is selected it drops connection until I power-cycle. 

This didn't happen until I had to reflash the ESP8266 to wipe wifi settings recently

ramapcsx2, 2022-08-05:
Many different mixes of source and output can cause this condition:
The WiFi signals are getting distorted by the activity of the video scaler chip, effectively blocking the network.
No good workarounds found yet, but know that it will work if the source video format changes, or the source turns off, etc.