ramapcsx2/gbs-control #354: Need to be able to lock output at around 60Hz (medium res)

f4f4f4f4, 2022-07-26:
Working on Sega Gunblade (medium resolution). It originally had a RGB projection TV, but now using an LCD TV. When GBS starts up, log says "source Hz: 57.560, output Hz:57.577"

The TV doesn't like 57.577Hz and displays "Unsupported". I noticed if I press the "60/50Hz PAL" button, the HTotal starts stepping down until the frequency hits 60.003, and the TV is fine with this and has a picture. I save the preset, but next time the game is turned on, output is back to 57.577 and the TV says "Unsupported".

Also tried manually stepping HTotal and then saving preset, to no avail.

JAOT, 2022-07-30:
Do you have the clock generator installed?
Converting 50 hz to 60hz behaves wrongly if you don't disable it.

JPman1aco, 2025-09-26:
same issue here. 
All my sega model 2 games output 57hz
I didn't found a definitive solution.

f4f4f4f4, 2025-09-26:
> Do you have the clock generator installed? Converting 50 hz to 60hz behaves wrongly if you don't disable it.

Apologies for not seeing this quick reply... but I'm not using an external clock gen.