ramapcsx2/gbs-control #356: Unable to get IIGS working coherently with latest GBSControl

Simon-Bitdiddle, 2022-07-30:
I am using a IIGS into a GBS8200, the only physical mod I am using is a 100 ohm resistor across the sync and ground pins. I have ensured that I am using the latest (7/29/2022) version of the software.

With the configurations completely reset, I am getting the following entries and then the ESP8266 resets.

...
<reset>
Scanning inputs for sources ...
Activity detected, input: RGB
VSync: present
HSync: present
RGB/HV
RGB/HV < > SOG

<reset>
...

What information can I provide to help troubleshoot this?

ramapcsx2, 2022-08-05:
The computer may output CSync + VSync. If so, try to not connect the VSync signal.

Simon-Bitdiddle, 2022-08-09:
I was able to solve my sync problems by connecting the debug line.

closed 2022-08-09 by Simon-Bitdiddle