ramapcsx2/gbs-control #361: Horizontal Oscillations on GBS Control (8200 with Clock Gen)

Razziel0, 2022-08-18:
Hello All,

I have assembled a GBS Control using a GBS 8200 with the following optional mods:

- Add 100ohm resistor across sync and ground for RGBs input
- Remove C11, optionally replace with 22uF (6.3V to 16V) electrolytic cap [I put a 22uf 16V cap]
- Remove the 3 RGB potentiometers, replace with direct link
- Add 10uf / 22uf ceramic SMD capacitors in parallel to stock ones (x4) C23, C41(alternative: C43), C42, C48

I am getting horizontal oscillations in the images when testing with my PS2 (using component cables) and my Sega Saturn (using HD Retrovision cable + adapter). 

I used a mix of the Wiki, Voular's, and Retro RGB's videos to assemble it. I'm an amateur to electronics and thought this was a fairly simple project considering the tutorials available. I'm able to log into the ad hoc network for the GBS Control and change settings but the oscillations are always present. the default resolution it picks is shown at the top of the album and the others are all worse (one example at the bottom). An album of Images and video of issue as well as my GBS unit linked below.

[Imgur Album](https://imgur.com/a/Cun9OvF)

Some troubleshooting steps I've taken:
- I am using shielded cables and tried generic non-shielded cables with no difference
- I tried connecting directly from Console -> GBS -> TV
- I reflowed the added ceramic capacitors to ensure a good connection
- I added the C11 22uf capacitor (still had issue before and after adding it)
- I moved the power wires for the clock gen board form C47 to C48 to ensure a good connection

Any help you can provide is sincerely appreciated as I'm lost for options for fixing this. I was hoping to use the GBS Control instead of having to try to get an expensive and at the time elusive RetroTINK 5X-Pro.

If you need any other information please let me know and I will try to provide it.

tunnotron3000, 2022-08-18:
Are you using PAL consoles by any chance? I get this when I chain my GBS into my OSSC and use my PAL PS1. Once I go 60Hz everything is fine. It also doesn't seem to happen when using a VGA to HDMI dongle.

Razziel0, 2022-08-19:
Hey, good idea. I checked both consoles and (un)fortunately they're both 60hz NA models. Though they both seem to have that issue. Have you encountered it from anything else? I'm using component in and component out into my TV's component ports, not sure if that would matter though.

tunnotron3000, 2022-08-19:
The latest firmware allows you to disable the clock gen, maybe try that and a reset? That helps eliminate the clock gen without having to faff about with desoldering. It could maybe be the power supply too, might be worth checking that it's up to spec. I do remember rama saying that the component out is experimental so you could also try switching to the VGA port?

If everything else fails, you may have a dodgy board.

ramapcsx2, 2022-08-19:
Hey, the first video looks like the new clock wasn't stable, but there could be countless reasons.
I'd also suggest you try to disable the clock gen and see whether it works 100% without it.