ramapcsx2/gbs-control #366: Help - Solder bridging while replacing R34 resistor

generalUrsas, 2022-08-27:
Please excuse me if this is the wrong place to post this request.

I am new to soldering but having a lot of fun with this project. I inadvertently bridged solder between R34 and C33. When I wicked the solder away, it appears that the two nodes are connected at the board level. The connection between seems to uniform to be organic.

Can someoner please advise on the best course of action and / or possiblly check for continuity across the two points in question on a known good board?

P.S. I did notice the foreign object between C32 and C31.

[image]

Thank you.

jsteel44, 2022-09-09:
I just checked mine. Where you have the small red circle between R34 and C33, I get continuity there.

generalUrsas, 2022-09-10:
Thank you very much for the reply. Since my original post, I have also had the opportunity to check against a brand new board. I also got continuity across the indicated points, and subsequently completed my build. Everything seems like it is working well.
Thanks again!

jsteel44, 2022-09-10:
Great, just out of interest, what guide are you following for your build? I don't remember having to touch the components pictured.

generalUrsas, 2022-09-10:
I opted for the alternate mod (75 Ohm R34 resistor instead of 100 Ohm across sync and ground) as listed on the wiki page under Hardware Mod library.

https://github.com/ramapcsx2/gbs-control/wiki/Hardware-Mod-Library

[image]

I should clarify. If I understand correctly, the above mod is only required for Scart input. So far, I have only used the component input, as I do not have any Scart devices. I only performed the mod for the practice and to sort of future proof the build. So when I say it is working, I really mean that the board remains partially untested, but seems to behave correctly for my present needs.