ramapcsx2/gbs-control #367: Can not connect to local Wifi - Stuck in AP Mode

knutwurst, 2022-09-04:
Hi @ramapcsx2,
Thank you for your work!

In the past I have used GBSControl on two devices. It always worked great.

Now some features and a new web interface were added, and I pulled the Updates, updated all my Arduino components and tried to compile.

However, now it seems like the wifi settings are not being applied. When I try to connect to any wifi (whether it's my home wifi, guest wifi or my cell phone's hotspot), I can't get a connection.

The SSID scan works fine, then I select the network, enter the (correct) password, click save and then only a reset happens and after that the GBS is back in AP mode.

I know that was not the case with older firmware.
I have already deleted all wifi settings and also tried to overwrite the entire flash - without success.
Another (new) Lolin WeMos D1 Mini didn't work either.

best regards,
Olli

Snocksman, 2022-09-06:
Same Problem here ! I wanted to Update my GBSControl to the latest Version and now I´m unable to connect it to my WiFi Network. I select STA/Scan Network, select my SSID, enter my WiFi Password and click the connect button. After that the WeMos D1 mini reboots, but is still stuck in AP-Mode.

best regards,
Mathias

Snocksman, 2022-09-13:
I found the Problem !  The sollution was described in the manual to GBS Control.

[https://ramapcsx2.github.io/gbs-control/Wiki/Software-Setup.html](url) <= Here you can find the hint: Go to Tools > Board: > Boards Manager ..., search for "esp8266" and install "esp8266 by ESP8266 Community",
version 2.6.3
Warning

    Do not use version 3.0.0 or newer, it is reported to break things!:

And yes, newer Version than 2.6.3 are really breaking things... like WiFi connection.

After Downgrading in the Arduino IDE Boardmanager to Version 2.6.3, the WiFi connection is working fine.

best regards,
Mathias

knutwurst, 2022-09-13:
To be honest I found the solution after some time, but forgot to close this ticket.

Thank you anyway! I hope this helps others :)

closed 2022-09-13 by knutwurst