ramapcsx2/gbs-control #368: [PROPOSAL]: Move WIKI to github pages

israpps, 2022-09-06:
the idea is to move the wiki into github pages, for that purpose, I made a draft sample, using the RUNDocs template

[See the webpage on action](https://israpps.github.io/gbs-control/Wiki/Build-the-Hardware.html)

[webpage source](https://github.com/israpps/gbs-control/tree/gh-pages)

Hope you like this Idea

ramapcsx2, 2022-09-06:
Hey, first of all, it looks amazing, cheers!

I don't have the time to manage gbs-control anymore, so please excuse me if this takes too much time on my side.
What needs to be done to move to the nicer system? :)

israpps, 2022-09-06:
> Hey, first of all, it looks amazing, cheers!
> 
> I don't have the time to manage gbs-control anymore, so please excuse me if this takes too much time on my side. What needs to be done to move to the nicer system? :)

1. Create a Branch called `gh-pages`
2. move the stuff i made into the `gh-pages` branch on the main repo
3. go to settings > pages and ensure github pages is enabled, on branch `gh-pages` and set on branch root, not a sub-folder
4. replace all references to the wiki with new ones to the corresponding github pages url

> I don't have the time to manage gbs-control anymore, so please excuse me if this takes too much time on my side.

What a shame. Real life tends to be demanding for everyone lol.

I bought the parts to make my first GBS-C, so far only the GBS Board arrived the country and i´m waiting for the rest of parts

I can help if u want. since 2020 i've became a member and developer of the PS2 community, my programming knoledge is still very basic, but at least I can help on other ways 😄 ...

ramapcsx2, 2022-09-06:
Thanks, really appreciate it :)

Don't let inexperience discourage you. 
We picks projects we like and learn the craft on them. Always been so ;p

Okay, so for this one, it is a hardware / firmware kind of project. It's heavy on the reverse engineering and understanding the TiVo scaler chip.
It got started because that hardware is very capable, but all the GBS products use minimum effort firmware. 
This just had to be fixed :)

You'll find that this is an Arduino project. Arduino was great for me at the time, but it's not well suited to this kind of task.
You'll also find that the code base is a huge, multi entangled mess. Sorry for that!
My excuse for this is that it just grew and grew, and I never made the time to properly rewrite it. I just kept adding things.

So that being said, maybe you can find some fun working on this. 
There's tons of backlog on the shmups forum thread, as well as all the older issues here on Github.

For the Wiki stuff, can you help me out a little and do a pull request?
I'd like to just merge your branch, and have all the new content in the right place :)

israpps, 2022-09-06:
> Thanks, really appreciate it :)
> 
> Don't let inexperience discourage you. We pick projects we like and learn the craft on them. Always been so ;p
> 
> Okay, so for this one, it is a hardware / firmware kind of project. It's heavy on the reverse engineering and understanding the TiVo scaler chip. It got started because that hardware is very capable, but all the GBS products use minimum effort firmware. This just had to be fixed :)
> 
> You'll find that this is an Arduino project. Arduino was great for me at the time, but it's not well suited to this kind of task. You'll also find that the code base is a huge, multi entangled mess. Sorry for that! My excuse for this is that it just grew and grew, and I never made the time to properly rewrite it. I just kept adding things.
> 
> So that being said, maybe you can find some fun working on this. There's tons of backlog on the shmups forum thread, as well as all the older issues here on Github.
> 
> For the Wiki stuff, can you help me out a little and do a pull request? I'd like to just merge your branch, and have all the new content in the right place :)

i'll be more than happy to help

In regards of the pull request, please, create a gh-pages branch, so i can make the PR into it

ramapcsx2, 2022-09-09:
Hm, I've created "gh-pages". Is that what you wanted?

ramapcsx2, 2022-09-10:
Okay, what's next? :)

israpps, 2022-09-10:
It's ready!

israpps, 2022-09-10:
[image]

israpps, 2022-09-10:
Now all references to the old wiki should be replaced I believe

https://ramapcsx2.github.io/gbs-control/

ramapcsx2, 2022-09-10:
Whoops, went overboard with the merge to master.. Sigh :p

israpps, 2022-09-10:
WTF

when I made the PR I picked both gh-pages branches...

ramapcsx2, 2022-09-10:
No worries, I'm rusty with git. Reverted the changes now, hope it's clean enough.

ramapcsx2, 2022-09-10:
Okay, so what went wrong and how do we avoid it for next attempt? :p

israpps, 2022-09-10:
Not sure how that happened

When I made the pull request I properly picked source and target...

You could download the files and push them to the correct branch if u want

ramapcsx2, 2022-09-10:
Oki, pushed all the new files, overwrote some old ones. The old wiki stuff seems broken now, unless you go to it via the Github menu.
I'm good with changing all mentions to "https://ramapcsx2.github.io/gbs-control/".
Thoughts?

israpps, 2022-09-10:
I just saw you fused wiki with master branch...

It's doable but makes the repo a chaos...

Also, I see the new wiki files made their way to the gh-pages branch, so IMO you should remove the wiki files from Master and configure the gh-pages branch as the github pages source

israpps, 2022-09-10:
Go to https://github.com/ramapcsx2/gbs-control/settings/pages

And make sure it looks like this
[image]

ramapcsx2, 2022-09-10:
Pages were already set up like that.
What files (or better entire folders) to delete?

israpps, 2022-09-10:
maybe you should revert back to commit feecc0df8a83245ccec5126147978254d0953d41