ramapcsx2/gbs-control #370: Cap placement - Power supply bypass

MrVeink, 2022-09-08:
[This page](https://github.com/ramapcsx2/gbs-control/wiki/Power-supply-bypass-capacitors) is quite unclear on which all points should one add the secondary 10uF to. In the text, they mention C23, C41, C42 and C48 but the image shows capacitors on C3, C6, C7, C42 and C48.

israpps, 2022-09-08:
Maybe numeration changes across GBS Models?

MrVeink, 2022-09-08:
> Maybe numeration changes across GBS Models?

But C23 and C41 do appear in the picture but don't have any extra capacitors.

MrVeink, 2022-09-08:
Also if that would be the case, they should mention that on different revisions it's in different spots or otherwise people would be putting caps in the wrong places