ramapcsx2/gbs-control #371: NEO GEO MVS MV2-F Sync Issue / Temp Fix

Rieper48, 2022-09-08:
So I have 2 MVS 2 Slots both with jittery sync with GBS control. I am using the following Impedance matching RGB & Sync PCB for NeoGeo (C)MVS https://oshpark.com/shared_projects/R8NJ9UQ4.

The only way I can get a stable picture is to use 15Khz mode, with syncwatcher off. I've only ever got it to sync once with an upscaled mode and syncwatcher off, then after a power reset it lost it. I sure hope these can be remedied for all MVS owners.

TioChico27, 2022-09-10:
I have a MV1F-Z here, working with no problem at all.

Rieper48, 2022-09-10:
Thats good. I wish both of mine did.

TioChico27, 2022-09-10:
Maybe it's a problem with that model in specific.

Rieper48, 2022-09-10:
I did yes and it made no difference unfortunatley. I do have an mv1f but it needs repairing. Would you be able to measure the voltage from your sync pin? Peak to peak? Maybe if i can tune the sync voltage it will work. Thanks a million.

TioChico27, 2022-09-10:
[image]

Rieper48, 2022-09-10:
Thank you!!! I'll compare with readings from my scope. Appreciate the help

Rieper48, 2022-09-11:
[image]

Kentsfield93, 2023-04-20:
Interesting, I've got the same problem here with my mvs mv-1fz. I've not yet installed the rgb bypass, which I will do once my pcbs and components arrive. I also learned about the dfo (dual frequency oscillator) project yesterday, which replaces the regular crystal on the mvs. The creator of this project mentioned, that he had some issues with his mvs and a xrgb / frame meister scaler as well. Seems to be a neo geo thing, I guess...

I'll update once I've installed the bypass + (maybe) the dfo mod.

Kentsfield93, 2023-05-03:
Quick update from my end - I got rid of all the problems by installing the DFO (Dual Frequency Oscillator) mod to my neo geo. Also, just for good measure, I installed the RGB mod to adapt the too hot TTL line levels from the NG to proper 0.6(ish) VPP ones. 

You can learn about the DFO mod here:
English: https://nfggames.com/forum2/index.php?topic=5744.0
German: https://circuit-board.de/forum/index.php/Thread/18016-DFO-Dual-Frequency-Oscillator/

You can either bild your own DFO but need to program it or you can source one. There are several shops in the US and Canada which sell these already pre-programmed. If you're within the EU, they sometimes get sold by members of some forums or otherwise Ebay. The seller Rugal_KOF94 has them in stock and ships from Spain (I bought mine from him, can recommend! He also has 5V versions but at the time of this comment, not listed - You need to send him a message via Ebay). You just need to make sure, if your NG uses 3.3V levels for it's XTAL line (like the MV1-C does) or 5V, like the MV1-FZ and FZS. Read through the forum posts to find out what your NG needs and how to measure it properly.

The frequencies I used for my MV1-FZS are the same, as for the MV1-C:
S0 = high => stock 24.00 MHz (= stock refresh rate 59.18Hz)
S1 = low => 24.3059 MHz (= refresh rate 59.94 Hz)

On the slightly overclocked setting, the sync issues are immediately gone and the picture looks perfect. Played for about 3 hours so far and ist just works without any problems.

The PCB for the RGB mod can be found here, including a BOM: https://github.com/philenotfound/cmvs-rgb

Hope it helps!

Aimen06, 2023-06-09:
I have tested with mv2F i have sync problem.
I have tested with  MV1-FZ it works great .
I have ordered a dfo for my mv2f , i hope it will work

Aimen06, 2023-06-20:
@Kentsfield93  Can you share the diagram for install  the dfo on a neogeo MVS MV2F  don't find it :/