ramapcsx2/gbs-control #375: Sync detection failure when using Composite/SVideo via C2C-64

KaixeRho, 2022-09-15:
How's it going.
To keep this as brief as possible: I was interested in getting composite and SVideo functionality working on the GBS-Control. The only transcoder for this was the Koryuu, but [as seen in this thread from a while ago](https://github.com/ramapcsx2/gbs-control/issues/142). I don't believe this issue was resolved. 

[I stumbled across the GitHub for the C2C-64](https://github.com/jsanpe/c2c-64) , and after much discussion with the lead dev, I was able to get my hands on a copy of the board. I was informed that the sync timings are identical to the Koryuu - if the Koryuu had sync detection failure issues, the C2C64 (basically a stripped down and minimalist version of the former) would likely have the same exact issues. Sure enough, it does have those issues. The image quality is excellent from source -> C2C -> GBSC, but the image is constantly shutting off after every frame. 

I'm not sure if it was a joke, but Rama remarked that a Koryuu model would need to be sent over to actually solve this issue. I really want to offer to do exactly that. I would love to send over my C2C-64. With both it and the GBS-C being open source, cheaper, DIY alternatives to things like the OSSC or the Koryuu, I think building compatibility between the two could only be a boon for this entire space. 

I understand that Rama specifically is busy with XStation, but I think if possible this might be worth pursuing, even if it takes a good amount of time.

jsanpe, 2022-09-15:
Hi all! I am the developer behind C2C. Just wanted to say hi, and participate in the discussion so we can try to find a way for interfacing C2C and GBS-control. In this regard, I would need to understand the sync signals that gbs-control expects and figure out a way to output them from C2C. If you have any documentation in this regard please share it here.

C2C uses the ADV7391/ADV7393 series of encoders for generating component/rgb signals. The datasheet is available [here](https://www.analog.com/media/en/technical-documentation/data-sheets/ADV7390_7391_7392_7393.pdf). There are some options for configuring sync signals, but it's not extremelly flexible. Hopefully we can make it work.

Cheers

ramapcsx2, 2022-09-15:
Hey, the amplitude on the ASIC sync input pins should be between 0.7Vpp and 3.3Vpp.
Sync failure is more about the not fully reverse engineered sync extractor / sync window settings, than any other thing.
Csyncs work better than separate syncs.