ramapcsx2/gbs-control #382: 1920x1080 incorect aspect ratio, too narrow, adding black bars

VicCatarino, 2022-10-08:
Hello, i am having a little problem with my GBScontrol (GBS-8220 Board), it happens when the output is set on 1920x1080, when using 4:3 consoles the aspect ratio on my TV appears too narrow (TV ration on 4:3), and when i set the ratio on my TV to 16:9 it appears to be correct but it looks like the GBS is adding black bars filling in the boarders of the screen (and for some reason the browser page stop working, it comes back when I disconnect the video input of the console, but it goes away as soon as I connect it again). 
N64 TV set to 4:3 too narrow, you can see the black bars the GBS is adding
[image]
N64 TV set to 16:9 aspect appears correct because of the black bars the GBS is adding to the image
[image]

The big problem is with widescreen consoles like the WII and PS2, over 1280x720 it works perfectily, although over 1920x1080 the ratio appear too narrow like being in 4:3 (and yes, the ratio config on my TV is correct in 16:9) and again it looks like the GBS is adding black bars filling in the boarders of the screen. 
Widescrenn WII TV set to 16:9 too narrow due to the black lines
[image]
Widescrenn WII TV set to 4:3 this way you can see the black lines in not from the TV
[image]

It started to happen about a mouth ago after i left my GBS of for some weeks, and to clarify I connected the GBS to other TV and to my monitor and got the same result. Did any of you have the same problem ? If yes what did you do ? do any of you have any solution ? 
I am thinking of remake the solder points an reinstall the firmware, I will post here if it works.

BTW english is not my native language, I am sorry for any grammatical errors or if i made myself hard to understand, it's been a while since i last practice

VicCatarino, 2022-10-10:
So, I re-uploaded the firmware two times and the same problem happens, I even cleaned the flash memory beforehand and still nothing.

JaCkBoston, 2022-11-28:
I may be wrong, and rama could correct me if he had the time to check on all of these but basically, this doesn't seem like an issue in and of itself. 

1920x1080p is inherently a 16:9 resolution. AFAIK, the way this works is as follows: GBS absolutely needs to output a full 16:9 frame to the screen in this mode in order for the monitor to understand it correctly. So it takes a frame of video in 4:3 or 2:3 or whatever the console outputs and adds rows of blank pixels to either side in order to fill the buffer and maintain that 16:9 aspect ratio that is inherent to the 1920x1080p signal. This is how this mode works. 

1080p is also not an integer scale of either of the common SD resolutions so I'd suggest you try using 960p instead, you won't lose anything in terms of picture quality (or at least, you shouldn't), this will fix your aspect ratio woes and while it will require some processing from your display, I don't think you'll experience anything noticeable in terms of processing time and input lag (that is, by you, a human, I'm sure something like an LDAT would notice at least some difference).