ramapcsx2/gbs-control #388: Fix the 1920x1080p Aspect Ratio please. And Maybe a fix for the VGA input?

R2MGaming, 2022-11-03:
I got a GBS Control from Ebay thinking it could work with any input whatsoever until I realized that yellow port was for the RGBs (thanks to a Reddit comment I received via my post) so I went with a Composite video and S-Video to VGA adapter, but it had caused some issues with the picture connected from composite to VGA. First issue is the picture being cropped partially to the right of the display (my TV), and the second issue is the 1080p preset not having an aspect ratio toggle too and maybe an improved picture toggle for VGA inputs.

https://www.dropbox.com/s/x85k9e1mdbx02v7/20221027_173924.jpg?dl=0 This is the one I got.

https://www.dropbox.com/s/fd4azvt2ao6502s/20221103_150454.jpg?dl=0 This is the issue I am receiving.

So to @ramapcsx2, Do you know of a good fix or solution, or even have one, yet have not uploaded it yet? I would love to update the firmware with this solution if you do.

R2MGaming, 2022-11-04:
Well I have messed around with the GBS Control I got on Ebay, but for some reason, the VGA input image gets scrambled for no apparent reason. Was there anything missing on the GBS Control, or do I need to mess around with the settings to get it to unscramble?

R2MGaming, 2022-11-04:
I've messed around with the GBS Control Web Ui with my device and for some reason, it just dislikes VGA set to 1080p perhaps. IDK why, but I'm gonna attempt to reupload the firmware to mine in a little bit.

R2MGaming, 2022-11-05:
I've just realized that with the composite and S-Video converter I got, all I have to do it allow Pass-through Mode one while the converter stays at 1080p. I will still keep this open for the time being when a new firmware for 1080p on the GBS Control is fixed.

ramapcsx2, 2022-11-05:
Hey, I'm sorry, but I don't have the time to develop things for gbscontrol anymore. I can barely check the issues from time to time :p

R2MGaming, 2022-11-05:
That is a bummer. Well do you know of anyone else who can still continue things for the GBS control?

R2MGaming, 2024-02-19:
I apologize for necro'ing this issue. I know you're still not working on this anymore, but would it at least be possible to add in a new option for the GBS C software to add in a 16:9 option along with the resolution options? I tried to tweak the 1080p resolution (which is still 4:3) by expanding the borders, scaling the display, and even moving it, but it always creates some screen tearing on the right, and when I move the display's position to the right, it always merges the display with the now-removed border making the picture uglier on the right. I've yet to show an example with my Wii and my TV (which didn't help at all) tomorrow.

ramapcsx2, 2024-02-19:
Hey,
you're hitting various limitations of the hardware. 
Especially the high line count presets are already maxing out all the parameters, in a way that just trying to do small adjustments often breaks one of the constraints rules. As a result you then get various artefacts.
It would be better if you could make the 1280x presets work for you. There's more room for tuning with those :)

R2MGaming, 2024-02-19:
Very well then. I'll use 1280 instead.

R2MGaming, 2024-02-19:
One more post before I close this issue: "I've been noticing some random fuzziness with a Wii Component cable connected to the GBS-C, and I would like to know if there is a workaround to that."

ramapcsx2, 2024-02-20:
The core ASIC on the GBS is ancient by now. It was designed around 2005/2006. It's a testament to the hardware that its as capable as it is, but it stands no chance against anything modern, yea :p

Random fuzziness is kind of vague :p

R2MGaming, 2024-02-20:
No wonder the GBS 82XX upscalers were cheap pieces of Chinese crap. It actually makes me wonder if the ASIC (I'm assuming it's the one with the heatsink to it) is the culprit with the terrible aspect ratio choices, resolution issues, or something else on the boards regardless if they were newly custom made ones with HDMI and SCART built in. I have the Mcbazel ODV GBS Control board so I wonder if an FPGA chip with the same pinout as the GBS core ASIC could work better than the ASIC, or would it also require more memory for it?
[image]

Metaln00b, 2024-02-20:
It's not a peace of crap but a good value for money and probably the cheapest hardware. It makes the video signal better than what the consoles can push out. The ODV-GBS-C from McBazel is in my opinion the best hardware for "plug and play".

The Retrotink 4K is very expensive and I dare to doubt that it gets a 4K resolution out of e.g. PS2 signals like a PS5.

R2MGaming, 2024-02-20:
> It's not a peace of crap but a good value for money and probably the cheapest hardware. It makes the video signal better than what the consoles can push out. The ODV-GBS-C from McBazel is in my opinion the best hardware for "plug and play".
> 
> The Retrotink 4K is very expensive.

I've thought of a better idea that could make the GBS-C better than the base modified units or even the custom made ones: instead of having FPGA to ASIC adapters, it could probably be possible to program some code to an official Arduino MKR Vidor 4000 (which has an Intel Cyclone 10 FPGA chip integrated into it (I got mine for $50 on Amazon back in 2022)) so... I am thinking this: "Why not have the Arduino MKR Vidor 4000 to override the current chips on the GBS Control board or work in conjunction with the current chips on the GBS-C?" Like let's say for example: "you could have the MKR Vidor 4000 to override the Aspect Ratio portion of the GBS-C and have the current chips on both the MKR Vidor 4000 and the GBS-C to double the resolution from 1080p60 to (call me crazy if you must) 4K60!" It would actually be a good cheaper alternative to the RetroTINK 4K, but it would be a little bit more expensive (but not expensive as the RT4K) to actually build too.

ramapcsx2, 2024-02-20:
Ehr yea, this whole GBSC project is about leveraging cheap hardware for a quite decent upscaler.
At the time, it compared well to alternatives, some of which were several times more expensive. 
It is now firmly an every man's upscaler :)

The problem here is that the 1080p presets are at the limit of the hardware, where compromises have been made to even have them.
The scaler is much more confident in 1280x960, where quality is pretty nice.

ramapcsx2, 2024-02-20:
@R2MGaming
What you imagine with the FPGA is one thing, whether any of this works or who might want to code such a project is another.
There's a good reason the Retrotink devices are quite costly. This development is extensive and full time.

R2MGaming, 2024-02-20:
> @R2MGaming What you imagine with the FPGA is one thing, whether any of this works or who might want to code such a project is another. There's a good reason the Retrotink devices are quite costly. This development is extensive and full time.

Well to be fair, before I thought of the MKR Vidor 4000 idea, I did think of the chip replacement idea (which was a bad idea to be fair), so I thought of the MKR Vidor 4000 idea since I do own one and IF there came one little project (I'm gonna be patient enough for one to happen one day) that could work in conjunction with the GBSC Project and have an option to override the current chips of the GBSC and/or combine both parts to enhance the resolution further could actually work.

Once again, it's just an idea that someone could adopt and make.

R2MGaming, 2024-02-20:
Follow-up to my reply: according to one Arduino forum (which I did start a post there to see if it was possible to make a modular upscaler that could rival the RetroTINK 5X and current GBS-C (which now is currently impossible to make since there wasn't any new shields that accept all kinds of legacy video inputs from RF coax to component video), but was not related to Arduino at all), @PhilippeAtSysEmb who responded to my post saying that it's possible to store video input to the MKR Vidor 4000's SDRAM chip making 1080p60 possible (if it could also work with adding in a 16:9 or 16:10 aspect ratio for the Arduino board) so... if it would be possible to actually collaborate with Philippe, maybe you and him along with the others who collaborated with you can make a NEW GBS Control repository with the added functionality to have the Arduino board to override and work with the chips on the GBSC boards (including the original GBS 82XX ones, the HDMI variants, and the custom made ones like my Mcbazel ODV variant which to my surprise is still update-able with your current firmware you made @ramapcsx2) to actually be able to rival the RetroTINK 4K upscalers, and make the MKR Vidor 4000 to add in a composite/S-Video decoder feature inside the FPGA to work with the green component port for composite video (including S-video with a S-video female to RCA male adapter) or with a composite and S-video to SCART adapter without having to spend an arm and a leg to buy a RT4K (which is still 1.5 times more expensive than a brand new PS5 or Xbox Series X console to be fair, 2.5 times way more expensive on Ebay). Anyways, if it could be possible, we would need to figure out which parts of the chips have the correct pins to solder wires (I might prefer dupoint wires for my MKR Vidor 4000 and use them again for a replacement IF mine breaks all of a sudden) and which other BOM parts could also work for the new project to go along with this.

ramapcsx2, 2024-02-20:
Oh you sweet.. xD
Let's just say, you forgot the part where someone actually needs to develop another FPGA scaler, a job that's a cool 6 months fulltime work, for pros.. :)

ramapcsx2, 2024-02-20:
It is honestly a specialist job. People who dedicate the required time and effort should do this, and we currently have a few different people on it.
They all sell the scaler you want, just with the price point that matches the work they put in :)

It was okay when I worked on GBSC, learning the trade and all. It was a side project and I was a student then.
I couldn't repeat the same effort today, but thankfully, others are already on it. :)

R2MGaming, 2024-02-20:
Okay...? Well do you know of anyone else who is willing to take the reigns? Also, about the FPGA scaling option for cheap, they could actually use the MKR Vidor 4000's FPGA chip (like I said: I bought my MKR Vidor 4000 for cheap on Amazon anyway) to override one of the functions for the GBS-82XX series.

YoImLee, 2024-02-20:
Seeing as you're the one who already has the hardware, why not have a crack at it yourself? It can't be that hard after all, surely.

R2MGaming, 2024-02-20:
> Seeing as you're the one who already has the hardware, why not have a crack at it yourself? It can't be that hard after all, surely.

Well the only problem for me is: "I have no programming knowledge with Arduino to be fair."

R2MGaming, 2024-02-20:
New post here, but IF someone WERE to adopt my idea for using the MKR Vidor 4000 along with the other parts of the GBSC boards, and program it to make picture better, and if there comes some video noise and random horizontal static (I should've explained better earlier), I might just get parts for another GBSC upscaler (mostly with an HDMI variant since those are better than VGA/RGBS to VGA only), and get some dupoint wires for my own. But it's like I said before: I have no programming knowledge of Arduino.

ramapcsx2, 2024-02-21:
Hm, I dunno whether I can explain it properly, but this doesn't at all work like you imagine it :p
It is not a question of pins or soldering, but rather what each processor in the system is designed to do, and then getting the most out of that processor.
As such, you cannot solder an FPGA board to the GBS ASIC and have that somehow enhance its internal capabilities.
The GBS ASIC has soandso SRAM bytes to store a video line, for example, and this can't be simply extended by slapping an extra chip on somewhere.
There are some things that can be done with an FPGA, but it is all much more complex than just a soldering job.
This is why I said it would take months to make anything worthwhile, and it probably would not even use the GBS at all then :p

R2MGaming, 2024-02-21:
Then I guess I may invest in a RetroTink 4K then I suppose.

ramapcsx2, 2024-02-21:
There's also the Morph from PixelFX, the OSSC or OSSC Pro or the various other Retrotinks. The 5X for example is fantastic, and should be a lot more affordable.
If it comes to price, check out the OSSC. For a good mix of price and features, check out the Morph.
These are all really good options that are professionally made :)

R2MGaming, 2024-02-21:
That's actually a good idea. The Morph 4K is cheaper, and has many different modules so I may scrap that idea I had.

R2MGaming, 2024-02-23:
Pardon this new post, but... I was Google searching stuff about the ASIC chip for the GBS boards (I haven't found anything about what the SDRAM looks like on there), and I saw a forum on arcade-museum.com and the user known as "DogP" actually bought himself a different converter with the Weststar WSC1115 IC (IF it is possible to make an 208 pin to 160 pin adapter if I'm assuming that the WSC1115 chip has 208 pins total compared to the Trueview 5725 ASIC's 160 pins) which I am assuming from one HTML (IF only there was an English translation of the the second page showing the chip) could actually be better than the Trueview 5725 and could actually work on any GBSC project board to remove video noise, remove, or reduce minor screen-tearing (when either motionless, or in motion which I've noticed as a bit of fuzzy static going from the bottom of my screen to the top and repeating itself) without messing with the clock generator (whether it's on a regular GBSC or a built in one on one PCB like the ODV-GBSC for example), work as a transcoder with composite and S-Video, and maybe even output 16:9 with 1080p60 for native widescreen system usage. If I speculated correctly, it could be possible to actually do an ASIC swap with the GBSC boards housing the Trueview chip and the converter's WSC1115 chip.

R2MGaming, 2024-02-23:
Okay, disregard the last post about the WSC1115 chip, I did some searching on Google to see if there was a better higher quality replacement to the Trueview 5725 ASIC and was trying to brainstorm what could work, but everything else turned out to be a bad idea until I came across a Xilinx FPGA (which has 160 pins and could be the same size as the Trueview5725) I had saved on my watchlist on Ebay (since FPGA's are reprogrammable), so I am thinking what could happen IF someone were to adopt your repo (I'm gonna root for @Aleksandr-S12 for that), and if there is a higher capacity SDRAM chip that is the same size and has the same pins as all GBS boards SDRAM chips, then I could probably see this become a better poor man's RetroTink 4K, OSSC Pro, and PixelFX Morph one day.

ramapcsx2, 2024-02-23:
It just does not work like this. You can't swap chips like they were any kind of equivalent. No pin amount means anything here, the various chips do vastly different things and nothing is compatible.
Let's stop here, I think the topic is done.

closed 2024-02-23 by R2MGaming