ramapcsx2/gbs-control #389: gbs-control: Will it work with RGBS output from Marconi Specrum Analyser?

brian-mk, 2022-11-04:
This is a question rather than an issue with GBS-Control...

I have a CRT based Marconi 2380 Spectrum Analyser that provides an RGBS output.
The composite sync output is: horiz: 15.625kHz, vert: 48.2Hz, non-interlaced. i.e. 324 lines.

I have tried a standard GBS-8200 (V4) to convert this to VGA.
Using default GBS-8200 auto settings I get a stable image that is slightly too large in the vertical direction i.e. some lines are missing from the top and bottom of the image.
If I try to reduce the vertical size using the menu buttons, the missing lines become visible but they appear to be out of sync (more so at the bottom).

(See example image below)

Should this 324 line input signal be expected to work with standard GBS-8200 firmware?
If not then if I add an ESP8266 with GBS-Control firmware, could I make this work?
Will it mean making changes to some of the TVIA 5725 registers to suit the 324 line input?

[image]

ramapcsx2, 2022-11-05:
I'd say give it a try. The line count may be odd (forgot the PAL specs.. 325 lines?), but the firmware will try to work with it :)

brian-mk, 2022-11-13:
I added the ESP8266 module to the GBS-8200 and successfully flashed the GBS-control firmware.

I connected the Spectrum Analyser RGBS output and fed the VGA into a laptop using OBS and a VGA->USB capture device. This is the same setup I used before with the stock GBS-8200. The OBS window remains completely black.

Here is the gbs-control developer mode diagnostic output after powering up the Spectrum Analyser.

====

Activity detected, input: RGB
.*****
no signal
h: 430 v: 647 PLL:0 A:7b7b7b S:07.10.00 H-   I:00 D:05e5 m:0 ht:   0 vt:   0 hpw:   0 u: 96 s: 0 S: 6 W:-71
****************************
no signal
h: 430 v: 647 PLL:0 A:7b7b7b S:07.10.00 H-   I:00 D:05e5 m:0 ht:   0 vt:   0 hpw:   0 u:384 s: 0 S:13 W:-74
************

====

It appears from the diagnostic output that the TVIA 5725 is trying to sync using 647 lines.
Based on the horiz & vert frequencies (15.635kHz & 48.2Hz, non-interlaced) , the analyser output is actually 324 or 325 lines.

Is there a way I can force GBS-control to sync to this signal - perhaps by modifying some registers in the TVIA 5725?

ramapcsx2, 2022-11-13:
GBSC tries to work out what the sync parameters are, and goes through a few variations, trying to lock onto something.
If nothing works out, the firmware isn't clever enough (or there are bugs).
I'd say try an older firmware for this. The format doesn't sound too tricky.

brian-mk, 2022-11-13:
Yes - I figured that out by looking at the source code.

There are a few things I don't understand about the diagnostics information:-

1)  What does 'no signal' actually mean when using GBSC?

With the original GBS-8200 firmware, this apparently indicates 'no composite sync' signal.
I know that composite sync is present and yet GBSC still displays 'no signal' when attempting to sync.

2) What does 'h: 430 v: 647' mean?
I assume it refers to the input resolution currently being used to attempt to sync.
A typical CRT monitor has a 4:3 aspect ratio with the horizontal dimension being wider than the vertical.
So 430, 647 would mean 647 dots horizontally and 430 lines vertically.
Logically (to me at least) this would be labelled v: 430 h: 647.
Yet the diagnostics output labels it the other way round i.e. h: 430 v: 647 ?

3) What do the remaining diagnostic values indicate?

ramapcsx2, 2022-11-14:
I'm sorry, replying to all of that properly is out of scope atm ><

Couple things though, maybe that gets you further:
The 'h: 430 v: 647' values are read out of the ASICs format detection unit. They are a hint to the software what it should attempt to get a sync lock, but they aren't a hard rule.
These values can and do get misdetected.
In this case, it looks like it sees double the actual video lines, and never seems to recover from that.
The problem is likely in the CSync separation units, such that the VSync is not correct.

When it says "no signal", translate that as "having trouble, working on it" ;p

GUI stuff should update, if the internal state is fine. With sync issues, that is not given. Some might work, some won't.

brian-mk, 2022-11-20:
I spent some considerable time going through both the GBSC source code and the 5725 Programming Guide and Register Definitions.
The 5725 documentation is appalling! A lot of information is sketchy or missing.
Hats off to ramapcsx2 for managing to figure any of it out.
I found some of the GBSC source code difficult to follow, so replaced some of the 'constants in code' with enums. This makes the code easier to understand - especially the video modes.

I wasn't able to figure out why the detected no of lines (v: 647) is double what it should be (324 or 325).
I found the same problem when I tried feeding a monochrome composite video output from the spectrum analyser into the GBS-8200 green RCA input socket.
I think there must be an issue with the internal 5725 sync separation.

Even if the sync separation worked correctly, It looks to me that the 5725 would not handle 325 input lines properly. It appears to be designed to only handle a few standard modes. The internal Mode Detection logic is not intended to work with an unknown input format such as this.

At this point I almost gave up.
However, I discovered that if I disabled the syncWatcher and applied the presets for PAL 240p using the diagnostics commands, I could get a stable image.
As with the original GBS-8200 firmware, the image appeared too large in the vertical direction.
However, I was able to adjust the vertical position and size without the extra lines appearing garbled (which is what happened in the original GBS-8200 firmware).

The output captured from OBS now appears as it should:-

[image]

All I need to do now is figure out the best way to modify the GBSC code to keep this as a default setting (including the v size & position parameters).

ramapcsx2, 2022-11-20:
In the very high register ranges, there are chunks allotted for the format detection unit.
If you can find out which of those the chip chooses for the input, you can try modifying that chunk, in an attempt to get the line count to the proper 324.
The chip *is* designed to take these kinds of formats, all of them. It's all about the software ;p

brian-mk, 2022-11-20:
There are some registers mssing from the TV5725 Registers Definition v1.1 PDF file (REG_S1_2B -> REG_S1_2D)

There are some definitions for the missing registers in the include file tv5725.h:-
    typedef UReg<0x01, 0x2B, 0, 7> GBS_PRESET_ID;
    typedef UReg<0x01, 0x2B, 7, 1> GBS_PRESET_CUSTOM;
    typedef UReg<0x01, 0x2C, 0, 1> GBS_OPTION_SCANLINES_ENABLED;
    typedef UReg<0x01, 0x2C, 1, 1> GBS_OPTION_SCALING_RGBHV;
    typedef UReg<0x01, 0x2C, 2, 1> GBS_OPTION_PALFORCED60_ENABLED;
    typedef UReg<0x01, 0x2C, 3, 1> GBS_RUNTIME_UNUSED_BIT;
    typedef UReg<0x01, 0x2C, 4, 1> GBS_RUNTIME_FTL_ADJUSTED;
    typedef UReg<0x01, 0x2D, 0, 8> GBS_PRESET_DISPLAY_CLOCK;

I am thinking that maybe I need to make use of the 'User Defined Mode' described in S1_80 and S1_81 to detect 325 lines.
S1_2B defined in TV5725.h includes bits for GBS_PRESET_ID and GBS_PRESET_CUSTOM but these are not described in TV5725 Registers Definition v1.1.

ramapcsx2, 2022-11-20:
Ah right, these are all empty space that I use for extra storage. I forgot about those.
It's register space that exists, but isn't used by the ASIC, so it's a neat trick to put some stuff there :)

"'User Defined Mode' described in S1_80 and S1_81" sounds right, but nearby should be all the specified modes. They are not set in stone, and can be modified the same as the custom slot.

brian-mk, 2022-11-20:
It looks like there could be a potential problem using S1_80 & S1_81 to detect user defined modes for low resolutions:-

For 325p lines & 15.625 kHz,
S1_80 (MD_USER_DEF_VCNTRL) needs to be set to (325 * 2) / 16 = 40.625 i.e. 0x28. That seems ok.
S1_81 (MD_USER_DEF_HCNTRL) needs to be set to 6750000 / 15625 = 432 i.e. 0x1B0. However the register is only 8 bits!

Unless the MD_USER_DEF_HCNTRL register can be ignored, that suggests user defined modes can only be used for higher resolutions?

ramapcsx2, 2022-11-21:
Honestly, I don't remember how that works..
But you don't need that custom slot. 
You want to try fixing whatever slot your device is supposed to be using, and it sounds a whole lot like the standard PAL one, allbeit with slightly low vertical refresh.

brian-mk, 2022-11-24:
I am still struggling to get the software do what I want.

Currently I have two issues:-

I loaded a preset (1280x1024). That results in a stable picture that is too large in the vertical direction. So I adjusted the vertical size & position and saved the configuration by creating a custom preset.
If I then load the saved preset, the size & position revert back to the original.
I don't understand what is going on here. Is the size & position information not saved?

Another issue is that having saved a preset, I can find no way to delete or rename it.

brian-mk, 2022-11-24:
I found a fault in getVideoMode(): The code I added to detect a new 'pal_324p' mode wasn't working correctly.
That had unexpected consequences.
Having fixed that, the system now appears to save and restore the position and size information correctly.
For some reason tho', the vertical and horizontal positions change slightly after power cycling. I have yet to work out why.

I discovered how to rename a preset - although not how to delete it.

There are some things I still do not understand...

1) Why are there doubled up preset IDs? For example 1280x960 exists as 0x1 and 0x11, 1280x1024 exists as 0x2 and 0x12 etc. The code checks for both ID values in many places.

2) I am confused by user created presets labelled as 'custom' vs presets labelled with a particular resolution such as '1280x1024'. Am I right in thinking that 'custom' refers to a saved preset that is not associated with a known videoMode?

ramapcsx2, 2022-11-24:
Iirc, the preset IDs double up for variations of the same source format. Was it like, 1280x960 for PAL and for NTSC? I think so.. :)

Sounds about right with the custom. Each preset requires a source format for it to work. Trying to apply something made for NTSC wouldn't work on PAL, etc.

brian-mk, 2022-12-01:
While attempting to detect the 324 line mode, one of the things I tried was to make use of the existing code in getVideomode() that detects a stable, but unrecognised, number of lines. It reads STATUS_SYNC_PROC_VTOTAL a couple of times in succession. If it appears stable (+/- 1 line) then it increments a 'notRecognized' counter, otherwise it resets the counter to 0. Either way it returns 'unknown mode' (=0). On subsequent calls to getVideomode(), if the 'stable' counter reaches 255 it returns 'unrecognised mode' (=15).

I thought this existing mechanism would be more flexible than adding code that expliciltly tests for 324 lines. It should be able to handle inputs with almost any number of lines (within reason).

What I found is that STATUS_SYNC_PROC_VTOTAL only gives a stable result (=323) when in sync.
When not in sync the register value varies considerably: The 'notRecognized' counter keeps being reset and never reaches 255. The mode remains set to 'unknown mode' (=0) and the system never manages to sync.

I ended up checking the VPERIOD_IF register instead. At least that appears to remain stable. It indicates twice the expected number of lines, so I have to test for 647 (+/- 1). Once in sync and the associated custom preset is loaded, if I enable 'print info' in the diagnostics, STATUS_SYNC_PROC_VTOTAL remains stable and always indicates 323. This method works but it's not as flexible as the existing 'unrecognised mode'.

Is an unstable STATUS_SYNC_PROC_VTOTAL when not in sync something you would expect?

ramapcsx2, 2022-12-01:
Yes, definitely! And welcome to the fun part of developing this ASIC firmware :D
I don't remember the details, but you basically have the major processing blocks that each can feed their data down the chain.
The top level for syncs is the sync processor, and it sees unfiltered line counts, all depending on the sync separation performance. When this is not stable, the firmware should try to tweak the sync separation or sync level parameters.
At the next stage, the sync processor feeds the created h/v syncs to the IF unit (input formatter), which prepares the digital processing. The IF has logic to filter the sync processor inputs, hence you see a stable signal here, but it can also lock into using an actually unstable signal and not notice it.

I'd say, use whatever works for you. If you can work with the IF line count, then go for it :)

brian-mk, 2022-12-02:
Years ago, I used to work for Crosfield Electronics as an embedded software engineer. One of the projects was a 'Colour Retouch Accelerator Board (CRAB)'  deigned to speed up desktop workstations so that they could compete with high end Crosfield Studio Systems.
That card also used an ASIC. The documementation provided by the ASIC designers was also 'poor' to say the least. Luckily we were able to talk to them directly to figure out how they expected the system to operate and ask about 'features' that were not written down.
Unfortunatley, getting answers that way is not posible with the 5725.

An example is the SP_DLT_REG.
The only information given in the registers definition document is:-
"Sync separation control. Sync pulse width difference threshold."
It doesn't tell you what that actually means, what effect it has or even which sync pulse (or pulses) it is refering to.
It doesn't tell you what the units are - is it no of clocks? (if so, which clock) or is it msec or usec?

I mention that one because I noticed that it gets adjusted in updateSpDynamic() when the video mode is 'unknown'.

    if (rto->videoStandardInput == 0 && vidModeReadout == 0) {
        if (GBS::SP_DLT_REG::read() > 0x30)
            GBS::SP_DLT_REG::write(0x30); // 5_35
        else
            GBS::SP_DLT_REG::write(0xC0);
        return;

vk5as, 2025-01-16:
@brian-mk I appear to be in a similar boat to you now, though not as familiar with the signals I am seeing.

I also have a 2380 which I am trying to convert into a modern video signal as the CRT has failed. Using the GBS-8200 I have a very unstable image that does not sync well. You mentioned you used a pull-up resistor to 3v3 for the sync signal, where exactly was this pull-up connected? See below image for my current signal present on the sync input pin to the GBS-8200 v4 board:

[image]

Given that the sync signal gets AC coupled, was the pull-up on the TIVA side of the coupling? Was there any other conditioning of the sync signal you did to improve things?

Also did you ever get to a stage with the software for it to be able to automatically detect the format?

brian-mk, 2025-02-05:
All that work I did to get the GBS-8200 to work with the 2380 Spectrum Analyser seems like a long time ago now.
I will turn 70 this year, my memory is not what it was and I don't remember alll the details.

What I can tell you is that the existing GBS-8200 software is unable to detect and handle the 2380 format.
I had to make extensive changes to the firmware to get it to work.
I would also say that the GBS-8200 source code is some of the worst I have ever encountered in some 40 years as a software engineer!

The guy who wrote it clearly understands video formats and how the ASIC works but doesn't have a clue about how to write or document code such that it can be easily read or modified by others.
In my case the 2380 CRT still works. I just wanted a way to provide hard copy. i.e. a graphics screen dump.

ramapcsx2, 2025-12-10:
Oh wow, I just read @brian-mk's comment, and I've got to say that I agree! :p
The source code is an example of how a personal use project grew and sprawled, later becoming open source, and no real cleanup applied whatsoever.
The software still does amazing things, which is ultimately what counts, but yea: It is not meant for others to work on ;p

"The guy who wrote it clearly understands video formats and how the ASIC works"
This is the thing, I do (or did) not :)
Everything in there has been found out experimentally, and over time, new findings made older code basically obsolete, yet that code wasn't refactored completely.
What you get in the end is what you see: Arduino, inexperienced developer, many contributions from various people, where everyone just added theirs on top... I hope it helps to see it all in a more fair way :)

But yea, it is proper horrible, got to agree :p

borzwatson, 2025-12-23:
> I was unable to find a way of uploading hidden files & folders (i.e. those with a dot prefix) using the web based git interface.
>
> I could only see a couple of those hidden files/folders i.e. .github/FUNDING.yml and .vscode/settings.json. They are missing from the repo.
>
> Rather than messing around installing WSL on my windows PC so I could use the git linux commands to get around the problem,
>
> I uploaded a zip file that has everything inside it (gbs-control.zip).
>
>
>
>
>
>
> Try this link:-
>
> GitHub - brian-mk/gbs-control: Modified version of ramapcsx2/gbs-control to work with Marconi 2830 Spectrum Analyser Display Unit.<https://github.com/brian-mk/gbs-control>
>
>
>
>
>
>
>
>
> Hi Brian,Yes, my address is ***@***.*** from my Galaxy
>
>
> I did wonder if there might be a problem trying to sending attachment in the reply.
>
> Can you provide a direct email address rather than going via github (ramapcsx2/gbs-control)?
>
> Brian.
>
>
>
>
>
>
>
>
> Hi Brian,
>
> Many thanks for looking this out for me.
>
> Unfortunately the zip file did not get through - please could you send a
>
> me link some other way (perhaps WeTransfer etc?)
>
> The picture I can download - very professional and neat build, an
>
> inspiration.
>
> Thanks again
>
> Regards,
>
> Simon

Hi Brian,Thanks again for uploading the code. If you can remember what the sync pull up resistor was please, all the other documentation about this shows a 100 R resistor down to 0v, but I appreciate it could be different with the 2380.A Merry Christmas to you and yours. ThanksRegards,SimonSent from my Galaxy

borzwatson, 2025-12-24:
> The composite sync signal from the 2830 is nominally 2V pk, ac coupled via a 75 Ohm series resistor.
>
> It requires a pull up resistor to 3.3V on the gbs8200 board. That's the orange wire in the photo I sent.
>
> I mounted the resistor inline (insulated with heat shrink sleeving).
>
> The resistor value is not critical as long as the input low voltage is pulled below 0.8V. I used 1K.

Thanks Brian, much appreciated.Regards,SimonSent from my Galaxy

borzwatson, 2026-01-15:
Hi @brian-mk 
The situation is that in passthrough mode  it all works well expect that the bottom and left edges of the display are chopped off on the VGA monitor. Because it is in passthrough mode, you cannot change the size of or move the screen in GBS.

If I switch to any of the other resolutions using the buttons on the GBS web interface, the VGA output becomes completely garbled.

As I mentioned a minor issue,  but the perfectionist in me would like to get all four edges of the display visible correctly.
Thanks again for all your help with this.

brian-mk, 2026-01-15:
I never tried connecting to a normal VGA monitor because I don't have one.
Perhaps that's why I never encountered your problem.
Instead I used a VGA to USB adapter together with OBS Studio running on a laptop.
That allows you to do frame grabs and get a hardcopy.

VGA video capture card: video capture supports input max. Resolution 1280 x 720 @ 60 Hz,
output max. Resolution 1920×1080@30Hz

[image]

brian-mk, 2026-01-16:
Something I noticed is that after selecting pass through mode, I a small red icon appears on the Web GUI:-

[image]

I could find no documentation that explains what this means.
However, after hunting around online, other users report that it mean loss of the WiFi gui connection.
That would explain why I was subsequently unable to disable pass through and had to power cycle the board to recover.

I have no idea why the WiFi connection should drop out after selecting pass through.
In my case I have gbs-control configured to connect to my home network via my router/hub rather than using the board as an access point. My router is in a different room so I guess the signal may be fairly weak. Perhaps selecting pass through mode creates some additional interference.  The antenna on the ESP module may not be that good. The WiFi connection from the router to my laptop sited near the board remains ok.

brian-mk, 2026-01-22:
[image]

[image]

borzwatson, 2026-02-10:
@ramapcsx2 from the screenshot above, any idea what may be incorrect? The picture is stable and doen't roll or flash, but it seems stretched out. If I turn on pass through, it works but of course you cannot scale the picture when in passthrough to make it fit. I include the timing detail when working in passthrough.

[image]

[image]

brian-mk, 2026-02-16:
Without spending a lot of time investigating, my guess is that you would need to look at making changes to the code snippet below specifically for VM_PAL_324p.

The diagnostic printVideoTimings() can be used to show the current horiz & vert scaling and position values.

Good luck.

====

    if (!isCustomPreset) {
        if (rto->videoStandardInput == VM_NTSC_480p || rto->videoStandardInput == VM_PAL_576p || (rto->videoStandardInput == VM_PAL_324p) ||
            rto->videoStandardInput == VM_HD_1080p || rto->videoStandardInput == VM_MED_RES_25kHz) {
            GBS::IF_LD_RAM_BYPS::write(1); // 1_0c 0 no LD, do this before setIfHblankParameters
        }

        //setIfHblankParameters();              // 1_0e, 1_18, 1_1a
        GBS::IF_INI_ST::write(0); // 16.08.19: don't calculate, use fixed to 0
        // the following sets a field offset that eliminates 240p content forced to 480i flicker
        //GBS::IF_INI_ST::write(GBS::PLLAD_MD::read() * 0.4261f);  // upper: * 0.4282f  lower: 0.424f

        GBS::IF_HS_INT_LPF_BYPS::write(0); // 1_02 2
        // 0 allows int/lpf for smoother scrolling with non-ideal scaling, also reduces jailbars and even noise
        // interpolation or lpf available, lpf looks better
        GBS::IF_HS_SEL_LPF::write(1);     // 1_02 1
        GBS::IF_HS_PSHIFT_BYPS::write(1); // 1_02 3 nonlinear scale phase shift bypass
        // 1_28 1 1:hbin generated write reset 0:line generated write reset
        GBS::IF_LD_WRST_SEL::write(1); // at 1 fixes output position regardless of 1_24
        //GBS::MADPT_Y_DELAY_UV_DELAY::write(0); // 2_17 default: 0 // don't overwrite

        GBS::SP_RT_HS_ST::write(0); // 5_49 // retiming hs ST, SP
        GBS::SP_RT_HS_SP::write(GBS::PLLAD_MD::read() * 0.93f);

        GBS::VDS_PK_LB_CORE::write(0); // 3_44 0-3 // 1 for anti source jailbars
        GBS::VDS_PK_LH_CORE::write(0); // 3_46 0-3 // 1 for anti source jailbars
        if (rto->presetID == PID_1920x1080_NTSC || rto->presetID == PID_1920x1080_PAL) {
            GBS::VDS_PK_LB_GAIN::write(0x16); // 3_45 // peaking HF
            GBS::VDS_PK_LH_GAIN::write(0x0A); // 3_47
        } else {
            GBS::VDS_PK_LB_GAIN::write(0x16); // 3_45
            GBS::VDS_PK_LH_GAIN::write(0x18); // 3_47
        }
        GBS::VDS_PK_VL_HL_SEL::write(0); // 3_43 0 if 1 then 3_45 HF almost no effect (coring 0xf9)
        GBS::VDS_PK_VL_HH_SEL::write(0); // 3_43 1

        GBS::VDS_STEP_GAIN::write(1); // step response, max 15 (VDS_STEP_DLY_CNTRL set in presets)

        // DAC filters / keep in presets for now
        //GBS::VDS_1ST_INT_BYPS::write(1); // disable RGB stage interpolator
        //GBS::VDS_2ND_INT_BYPS::write(1); // disable YUV stage interpolator

        // most cases will use osr 2
        setOverSampleRatio(2, true); // prepare only = true

        // full height option
        if (uopt->wantFullHeight) {
            if (videoModeIsNtsc(rto->videoStandardInput)) {
                //if (rto->presetID == PID_1920x1080_NTSC)
                //{ // out 1080p 60
                //  GBS::VDS_DIS_VB_ST::write(GBS::VDS_VSYNC_RST::read() - 1);
                //  GBS::VDS_DIS_VB_SP::write(42);
                //  GBS::VDS_VB_SP::write(42 - 10); // is VDS_DIS_VB_SP - 10 = 32 // watch for vblank overflow (ps3)
                //  GBS::VDS_VSCALE::write(455);
                //  GBS::IF_VB_SP::write(GBS::IF_VB_SP::read() + 4);
                //  GBS::IF_VB_ST::write(GBS::IF_VB_ST::read() + 4);
                //  SerialM.println(F("full height"));
                //}
            } else if (videoModeIsPal(rto->videoStandardInput)) {
                if (rto->presetID == PID_1920x1080_PAL) { // out 1080p 50
                    GBS::VDS_VSCALE::write(455);
                    GBS::VDS_DIS_VB_ST::write(GBS::VDS_VSYNC_RST::read()); // full = 1125 of 1125
                    GBS::VDS_DIS_VB_SP::write(42);
[... 158 lines cut ...]
            SerialM.println(pllRate);
            if (pllRate > 200) {             // is PLL even working?
                if (pllRate < 1800) {        // rate very low?
                    GBS::PLLAD_FS::write(0); // then low gain
                }
            }
            GBS::PLLAD_ICP::write(6); // all 25khz submodes have more lines than NTSC
            GBS::ADC_FLTR::write(1);  // 5_03
            GBS::IF_HB_ST::write(30); // 1_10; magic number
            //GBS::IF_HB_ST2::write(0x60);  // 1_18
            //GBS::IF_HB_SP2::write(0x88);  // 1_1a
            GBS::IF_HBIN_SP::write(0x60); // 1_26 works for all output presets
            if (rto->presetID == PID_1280x960_NTSC) {   // out x960
                GBS::VDS_VSCALE::write(410);
            } else if (rto->presetID == PID_1280x1024_NTSC) { // out x1024
                GBS::VDS_VSCALE::write(402);
            } else if (rto->presetID == PID_1280x720_NTSC) { // out 720p
                GBS::VDS_VSCALE::write(546);
            } else if (rto->presetID == PID_1920x1080_NTSC) { // out 1080p
                GBS::VDS_VSCALE::write(400);
            }
        }
    }

====