ramapcsx2/gbs-control #392: Leningrad 48K (ZX Spectrum Clone) Won't Sync.

genetoye, 2022-11-20:
The Leningrad outputs RGBS and worked successfully with a stock GBS 8200 V4.  After upgrading the board to gbscontrol with external clock, it will not sync the image.  It switches between no sync and partially sync'ed.  Should this work with gbscontrol?  Attached are images of the output and the sync signal.  The debug console repeats the following:

.** (debug) ign. length: 0x75
.* (debug) ign. length: 0x81
. (debug) ign. length: 0x90
.* (debug) ign. length: 0x7E
.** (debug) ign. length: 0x75
.** (debug) ign. length: 0x75

Here is the video timings:
HT / scale   : 2032 585
HS ST/SP     : 0 132
HB ST/SP(d)  : 2004 360
HB ST/SP     : 1944 222
------
VT / scale   : 1066 512
VS ST/SP     : 1 6
VB ST/SP(d)  : 1064 23
VB ST/SP     : 3 20
IF VB ST/SP  : 44 46
CsVT         : 312
CsVS_ST/SP   : 4 1
 (debug) ign. length: 0x90

.** (debug) ign. length: 0x75

.** (debug) ign. length: 0x75
[image]

.* (debug) ign. length: 0x75

.* (debug) ign. length: 0x75
[image]

.
[image]

Miron-666, 2023-04-12:
Hello! I have a Pentagon-128 and have the same problem. I solved the problem through a crutch. We change line 4844 in the gbs-control.ino file from rto->coastPositionIsSet = 0 to rto->coastPositionIsSet = 1. After turning on the computer, I wait for a stable image in the menu and then disable the syncwatcher in the web interface. After that, everything works fine, including demos with multicolor. Maybe there is a more elegant solution, but I'm not good at it. Sorry for my English, I use a translator.

Miron-666, 2023-04-12:
Video timings are as follows:

HT/scale: 1440 1023
HS ST/SP : 856
HB ST/SP(d) : 1260 254
HB ST/SP : 1202 168
------
VT/scale: 1125455
VS ST/SP : 16
VB ST/SP(d) : 1125 42
VB ST/SP : 3 16
IF VB ST/SP : 60 62
CsVT: 317
CsVS_ST/SP : 4 1

Miron-666, 2023-04-14:
Rejoiced early. When changing the color of the border to red and purple, there is a breakdown in synchronization and the screen jumps. Most likely there will be problems when loading from the tape when stripes move across the screen. I use a disk drive and did not notice much discomfort.

Telecrab, 2024-06-22:
Bumping this because having the same issue with another Spectrum clone (ZX-777).

Timings

```
HT / scale   : 2704 512
HS ST/SP     : 8 160
HB ST/SP(d)  : 2594 612
HB ST/SP     : 2512 480
------
VT / scale   : 1000 410
VS ST/SP     : 1 4
VB ST/SP(d)  : 1000 38
VB ST/SP     : 24 26
IF VB ST/SP  : 6 8
CsVT         : 296
CsVS_ST/SP   : 14 11
```
Here's "print info" output if this of any help.
```
h: 446 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 126 u:  a s: 0 S:13 W:31
h: 446 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2345 vt: 295 hpw: 126 u:  b s: 0 S:13 W:31
h:   5 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:02 D:048c m:0 ht:2325 vt: 295 hpw: 126 u:  b s: 0 S:13 W:31
h:   5 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2325 vt: 296 hpw:  99 u:  c s: 0 S:13 W:31
h: 446 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0400 m:0 ht:2345 vt: 295 hpw: 126 u:  d s: 0 S:13 W:31
h: 511 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 125 u:  e s: 0 S:13 W:31
h: 446 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0400 m:0 ht:2345 vt: 295 hpw: 133 u:  e s: 0 S:13 W:31
h: 442 v: 478 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2306 vt: 295 hpw: 126 u:  f s: 0 S:13 W:31
h: 511 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2306 vt: 296 hpw: 126 u: 10 s: 0 S:13 W:31
h: 446 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 126 u: 11 s: 0 S:13 W:31
h: 446 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 126 u: 12 s: 0 S:13 W:31
h: 446 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 125 u: 12 s: 0 S:13 W:31
h: 440 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2345 vt: 295 hpw: 128 u: 13 s: 0 S:13 W:31
h: 446 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:02 D:0484 m:0 ht:2345 vt: 296 hpw: 126 u: 14 s: 0 S:13 W:31
h: 446 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 126 u: 15 s: 0 S:13 W:31
h: 446 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 127 u: 16 s: 0 S:13 W:31
h: 436 v:1599 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2345 vt: 295 hpw: 129 u: 17 s: 0 S:13 W:31
h: 436 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 296 hpw: 126 u: 18 s: 0 S:13 W:31
h: 446 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw: 126 u: 19 s: 0 S:13 W:31
h: 446 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2345 vt: 295 hpw:  99 u: 1a s: 0 S:13 W:31
*h:   7 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:02 D:0484 m:0 ht:   5 vt: 295 hpw: 126 u: 1b s: 0 S:13 W:31
h: 446 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2505 vt: 296 hpw: 119 u: 1c s: 0 S:13 W:31
h:   5 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:02 D:0484 m:0 ht:2505 vt: 296 hpw: 126 u: 1c s: 0 S:13 W:31
h:   5 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2505 vt: 295 hpw: 126 u: 1d s: 0 S:13 W:31
h:   5 v:   4 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2505 vt: 295 hpw: 126 u: 1e s: 0 S:13 W:31
h:   5 v:   4 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:2505 vt: 295 hpw: 126 u: 1e s: 0 S:13 W:31
h: 446 v: 813 PLL:1 A:ffffff S:02.18.20 H-   I:02 D:0408 m:0 ht:2345 vt: 295 hpw: 126 u: 1f s: 0 S:13 W:31
h:   5 v: 813 PLL:0 A:ffffff S:02.18.20 H-   I:02 D:0484 m:0 ht: 320 vt: 296 hpw: 126 u: 20 s: 0 S:13 W:31
h:   5 v: 813 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 320 vt: 295 hpw: 126 u: 21 s: 0 S:13 W:31
h:   5 v: 813 PLL:0 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 320 vt: 295 hpw: 127 u: 22 s: 0 S:13 W:31
h:   5 v: 813 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 320 vt: 295 hpw: 126 u: 22 s: 0 S:13 W:31
h: 446 v: 813 PLL:1 A:ffffff S:02.18.20 H-   I:02 D:048c m:0 ht:2344 vt: 295 hpw: 126 u: 23 s: 0 S:13 W:31
h:   5 v: 813 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 349 vt: 296 hpw: 126 u: 24 s: 0 S:13 W:31
h:   5 v: 813 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 349 vt: 295 hpw: 126 u: 25 s: 0 S:13 W:31
h:   5 v: 289 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 349 vt: 295 hpw: 126 u: 26 s: 0 S:13 W:31
h:   5 v: 289 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 349 vt: 295 hpw: 126 u: 26 s: 0 S:13 W:31
h: 446 v: 289 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2344 vt: 295 hpw: 126 u: 27 s: 0 S:13 W:31
h:   5 v: 289 PLL:1 A:ffffff S:02.18.20 H-   I:02 D:0484 m:0 ht: 381 vt: 296 hpw: 126 u: 27 s: 0 S:13 W:31
h:   5 v: 289 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 381 vt: 295 hpw: 126 u: 28 s: 0 S:13 W:31
h:   5 v: 289 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 381 vt: 295 hpw: 126 u: 29 s: 0 S:13 W:31
h:   5 v: 289 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht: 381 vt: 295 hpw: 126 u: 2a s: 0 S:13 W:31
h: 511 v: 370 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2344 vt: 296 hpw: 127 u: 2b s: 0 S:13 W:31
h: 446 v:1279 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:048c m:0 ht:2343 vt: 297 hpw: 127 u: 2c s: 0 S:13 W:31
h:   5 v:1279 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:1792 vt: 296 hpw: 126 u: 2d s: 0 S:13 W:31
h:   5 v:1279 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:1792 vt: 295 hpw: 126 u: 2e s: 0 S:13 W:31
h:   5 v:1279 PLL:1 A:ffffff S:02.18.20 H-   I:00 D:0484 m:0 ht:1792 vt: 295 hpw: 126 u: 2f s: 0 S:13 W:31
h:   5 v:1279 PLL:0 A:ffffff S:02.18.30 H-   I:00 D:048c m:0 ht:2345 vt: 295 hpw: 126 u: 30 s: 0 S:13 W:31
[... cut ...]
h: 446 v: 623 PLL:0 A:ffffff S:02.18.20 H-   I:04 D:058c m:0 ht:2482 vt: 295 hpw: 134 u: 40 s: 0 S:13 W:31
h: 446 v: 623 PLL:0 A:ffffff S:02.18.20 H-   I:04 D:0508 m:0 ht:2483 vt: 296 hpw: 133 u: 41 s: 0 S:13 W:31
h: 446 v: 623 PLL:0 A:ffffff S:02.18.20 H-   I:04 D:05a4 m:0 ht:2481 vt: 294 hpw: 133 u: 42 s: 0 S:13 W:31
h: 446 v: 623 PLL:0 A:ffffff S:02.18.20 H-   I:04 D:058c m:0 ht:2482 vt: 295 hpw: 134 u: 43 s: 0 S:13 W:31
h: 446 v: 623 PLL:0 A:ffffff S:a7.10.20 H-   I:44 D:058c m:2 ht:2482 vt: 296 hpw: 135 u:  0 s: 1 S:13 W:31
```

Telecrab, 2025-03-18:
@Miron-666, @genetoye, have you had any success with this issue?

Miron-666, 2025-03-19:
> [@Miron-666](https://github.com/Miron-666), [@genetoye](https://github.com/genetoye), have you had any success with this issue?

Unfortunately, I was not able to solve this problem. This is a software problem and I cannot solve this task without the necessary knowledge. The problem is in the non-standard frequency of my Pentagon-128 computer. Frame rate is 49.87 Hz and line rate is 15.96 kHz (with a 14 MHz quartz resonator, frame rate is 48.828 Hz and line rate is 15.625 kHz). I decided to use another adapter on the Raspberry Pi Pico: https://zx-pk.ru/threads/36013-rgb-vga-hdmi-adapter-ot-aleksekb.html