ramapcsx2/gbs-control #393: Unable to Upscale on Consoles

Granblue1982, 2022-11-22:
Hello everyone.  After building my GBS-C out of an 8200, adding a 100 ohms 0.5 resistor, adding the clock generator, and trying it out, I'm running into a problem.

I have a PS2 Slim hooked up to the GBS-C via HD Retrovision component cables (The RGB Cables).  Using this in passthrough, 15Khz or, 480p/576p is fine.  However, when I go to upscale to any other higher resolution I run into an issue of a blank screen/No Signal on any other resolution.

The output I get when I have the console on is listed below.

```
ADC offset: R:40 G:40 B:40
clock gen reset: 81000000
HTotal Adjust: -5

preset applied: 720x480 for NTSC 60Hz 

source Hz: 59.94045 new out: 59.94049 clock: 80986464 (-13536)
ADC offset: R:40 G:40 B:40
clock gen reset: 108000000
HTotal Adjust: -4

preset applied: 1280x720 for NTSC 60Hz 

source Hz: 59.94162 new out: 59.94166 clock: 107979528 (-20472)
ADC offset: R:40 G:40 B:40
clock gen reset: 162000000
HTotal Adjust: -5

preset applied: 1280x960 for NTSC 60Hz 

source Hz: 60.28336 new out: 60.28341 clock: 162906288 (+906288)
matched preset override > 1280x960
ADC offset: R:40 G:40 B:40
clock gen reset: 162000000
HTotal Adjust: -5

preset applied: 1280x960 for NTSC 60Hz 

source Hz: 59.74429 new out: 59.74434 clock: 161449648 (-550352)
matched preset override > 1280x960
ADC offset: R:40 G:40 B:40
clock gen reset: 162000000
HTotal Adjust: -5

preset applied: 1280x960 for NTSC 60Hz 

source Hz: 59.92150 new out: 59.92150 clock: 161928416 (-71584)
ADC offset: R:40 G:40 B:40
clock gen reset: 108000000
HTotal Adjust: -3

preset applied: 1920x1080 for NTSC 60Hz (set your TV aspect ratio to 16:9!)

source Hz: 60.13094 new out: 60.13094 clock: 108249856 (+249856)
```

My question is if there is something I might be missing in terms of having a setting on, or is there something I might be missing in regards to modding my board?

Thanks to anyone who can help.

ramapcsx2, 2022-11-22:
Try a different power supply. This almost looks like instability from not enough juice :p

Granblue1982, 2022-11-23:
Had a stupid moment and jumped in voltage in regards to my power supply, ended up possibly burning out a board, but for sure I blew up a VGA to HDMI adapter.  Managed to wire up to another board with a different power supply and connect it to a PC monitor to see if that would work.

Was able to switch resolutions and everything else.  I'm hoping this works when the new adapter and parts come. 

Thanks again!

alextwothousand, 2022-12-09:
I have mine modded for USB-C, might be worth doing that, that way even if you use a 20V/100W charger, you can still limit it to 5V through PD negotiation.