ramapcsx2/gbs-control #394: VHS SCART to GBS (blank screen)

MaxJumper, 2022-12-04:
Hello!
I have a GBS8200 and I installed GBS-C there
I also made a modification of RGB
But the problem is that if everything works with PS2 (in RGB mode).
Then when I connect the Panasonic nv-hs930ee S-VHS, there is only a black screen on the screen. I.e. there is synchronization, but there is no picture.
And I don't understand how I can achieve the image. Maybe I should give permission signals on START to select the mode?

scheme [https://disk.yandex.ru/i/50ab53Z3_d6RLA](url)

ramapcsx2, 2022-12-04:
"S-VHS"?
The GBS hardware can decode RGB and Component signals only.
This sounds like S-Video, which you can plug into the green Component input as a test. If you get a black and white image there, it is S-Video, and not supported by the hardware.

MaxJumper, 2022-12-05:
https://disk.yandex.ru/i/QK1KYCcKcPd3-w
not S-Video
must be Scart RGB...

santiis2010, 2023-02-09:
> https://disk.yandex.ru/i/QK1KYCcKcPd3-w not S-Video must be Scart RGB...

for what i know about VCR's they only can output Composite and RF only if they are VHS, and RF, Composite and S-Video only if they are S-VHS, there isn't any VCR with RGB output since VHS can't output RGB for what i know.

My guess is that SCART output is only composite video out since thats the nature of VHS players, just use the S-Video output my friend.