ramapcsx2/gbs-control #395: Passthrough with scanlines

rafaelsb, 2022-12-05:
Hello, 

I think im having a problem with passthrough mode. Whenever I use it, the video is displayed at the original resolution but with scanlines (thick horizontal lines).

I dont think this is a normal behaviour and I can't disable the scanlines no matter what.

I've tested my Wii(480p source) and PS2(480i source) via component input and HDMI output in 2 different TVs, but the result is the same.

Is this normal or am I missing something?

Thx in advance

filevans, 2023-04-12:
which model of gbs control are you using?

risingPhil, 2025-07-08:
I can confirm this issue on this model from ticknot store:

[image]

I bought the scaler in february 2024.

Note: mine has s-video and composite, but the current ones being sold have an hdmi input instead.

I have no clue what commit version the firmware was based on. But I did upload its firmware to archive.org some time ago: https://archive.org/details/gbscontrol_with_composite_and_svideo_original_firmware.

[image]

I currently have my PS2 connected to SCART, output set to 480p, motion adaptive deinterlacing is turned on and scanline options look like this:

[image]

However, like OP, I can't turn them off when 480p is selected as output. However, I have no scanlines when I select a different output type (1080p, 720p, ...)

For me, it kinda works out because I wanted to test PS2 with scanlines and this gives me the option, however, it would be better if I could control it. (or if scanlines would be possible at any output resolution if the input is 480p)