ramapcsx2/gbs-control #396: GBS board not responging - Input source not recognized using ....  - Retroupgrades guide wrong ?

miguelk5, 2022-12-09:
Hello, I think that this project is not ready for not electronic enginers is nothing but a BETA and it should be comercial anymore. I had the kit from retroupgrades: https://www.retroupgrades.co.uk/guides/gbs-c-aio-kit-installation-guide/

1 - When you download this git repository before importing this project to arduino ide you need to remove the master from the folder name.

2- You have to attempt like 10 / 20 times before you can flash the ESP module: 

I only got this error hundred of times: esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

Until finally it worked:
Wrote 715184 bytes (526744 compressed) at 0x00000000 in 12.5 seconds (effective 457.5 kbit/s)...
Hash of data verified.

It can be so albitrary, at least give a warning about it in the doc, also you do not ever mention download necessary ESP module drivers in my case: https://raw.githubusercontent.com/nodemcu/nodemcu-devkit/master/Drivers/CH341SER_WINDOWS.zip

I had used this ESP: WeMos D1 NodeMcu Lua : https://m.media-amazon.com/images/I/616jAFl4j7L._AC_SL1001_.jpg

3- Manual installation of libraries doesn't work like you mention on the doc:  at your documentation you say that you need to download these 2 projects and place them in to libraries folder:

https://ramapcsx2.github.io/gbs-control/Wiki/Software-Setup.html#espasynctcp-and-espasyncwebserver-by-me-no-dev

I just got an error after doing that, I had imported the library as a ZIP and just pickup the zip file downloaded from the repos.

4-  In the retroupgrades guide I do not see refferences to the jumper

Here I can see a jumper: https://i.imgur.com/9DciLIe.png

https://ramapcsx2.github.io/gbs-control/Wiki/Build-the-Hardware.html

I believe that its necessary they do not mention to at least checkout the original documentation. should I put or not the jumper ?

5- The control panel navigation its so weird and bugged: when I navigate over the GBScontrol web gui it doesn't respond to my actions I have to click the buttons many times, the button reset values doesn't do anything, the option reset chip from the dev mode breaks the unit and after that I just see GBS board not responging.

6- It doesn't output anything from SCART to hdmi: I enabled the console, and saw something like: Input source not recognized using .... 

Eventually after change the switchs to RGBHV and the STRIP apparently was not touching none of the positions, so I had forced a little to ON so I could stop getting that error and saw this: input source component, but in that time I had pressed the option: reset chip (at the developer mode) and then I just got all the times GBS board not responging when tried to do any action thing that before didn't happened.

So I tried to flash again the unit but now I just get a timeout when upload, and I'm tired of  trying again and again.

Its great to develop projects but in my opinion this is just a concept if you're going to release something to the public at least make the documentation consistent, and the web gui I believe that is bugged.

I had trusted this project now I just wasted a lot of time and money for nothing .

miguelk5, 2022-12-09:
Please consider this feedback to do improvements, apparently its a fault of the retroupgrades PLCC socket, however I spent like 130€  and many time already so I will stop working on this, and purchase a comercial version: https://es.aliexpress.com/item/1005004167274730.html?pdp_npi=2%40dis%21EUR%21%E2%82%AC%20120%2C91%2185%2C85%20%E2%82%AC%21%21%21%21%21%40211b423d16706049190354152eb635%2112000028899615899%21btf&_t=pvid%3A4dfe735d-8680-48c0-b1d7-a7676b582d4c&afTraceInfo=1005004167274730__pc__pcBridgePPC__xxxxxx__1670604919&spm=a2g0o.ppclist.product.mainProduct&gatewayAdapt=glo2esp

ramapcsx2, 2022-12-09:
Hey, I'm sorry you had all these troubles, but GBSC the project is not commercial.
Several groups have used the software to offer something more refined than a bare board, but you'll have to reach out to them for support on your purchases.
This project just provides the software as is :)

WacKEDmaN, 2023-04-24:
i had same issues trying to flash a brand new "wemos d1" clone... first it wouldnt connect and reboot into flash mode like it should...i got around that, by forcing it into flash mode b4 power it up.... but no matter what i did i kept getting checksum errors... 
turns out it was a bad board (more specifically bad flash).. made up my own ESP8266 module on a breadboard that worked perfect first go...
seems the newer "clones" of the wemos d1 are complete garbage!