ramapcsx2/gbs-control #398: No output using VGA/HDMI converter at 50Hz?

lovrenco, 2022-12-18:
I tried from the PAL Amiga. Without converter it works normally on VGA momitor.
But with hdmi converter black screen 1920@50p
Since on Amiga you have early boot screen you can switch to NTSC by pressing SPACE, i dit that on blind, and it showed the picture 1920@60p

Any idea why? I tried EDTV force to 60, no luck.

lovrenco, 2022-12-23:
To clarify, question is more related to converter, than gbs-control itself. 
I want to know if someone other have same experience with that and is there any workaround?

DerpyChap, 2022-12-27:
I had this issue before, it's a problem with the converter not supporting PAL refresh rates. The following converter seems to support PAL refresh rates without any issues: https://www.amazon.co.uk/gp/product/B072NCTBVP/

Keep in mind I tested using a PS2, so your mileage may vary. I've found that 1080i passthrough doesn't seem to work with it (1080p signals are fine), although that's not much of a concern in my use case.

lovrenco, 2022-12-31:
> I had this issue before, it's a problem with the converter not supporting PAL refresh rates. The following converter seems to support PAL refresh rates without any issues: [...]

Thanks. I think I found same model on Aliexpress. I might try just that:
https://a.aliexpress.com/_msAXxPC

M-Reimer, 2023-01-31:
@lovrenco Can you update this Issue and maybe close it if the new adapter worked?

closed 2024-10-29 by ramapcsx2